gqrx-sdr/gqrx #223: Add .dmg build for latest version (2.3.x)

techniq, 2015-01-15:
It appears the last Mac OSX .dmg build was for version 2.2.0.  I was going to update the [homebrew cask](https://github.com/caskroom/homebrew-cask/blob/master/Casks/gqrx.rb) to the latest version and I noticed there hasn't been a build since 2.2.0.  Thanks.

csete, 2015-01-15:
I can not provide usable .dmg file and requesting it here is not going to make any difference whatsoever, sorry. Please see the very recent discussions on the mailing list.

csete, 2015-01-15:
By the way, I don't understand what homebrew cask is. There is an up to date recipe for macports (stable and HEAD) and also for homebrew. Neither of those require dmg, but could perhaps be used to create one?

closed 2015-01-15 by csete

techniq, 2015-01-15:
Understood.  I originally looked to install gqrx via homebrew but didn't see an official recipe (I found [chleggett/homebrew-gqrx](https://github.com/chleggett/homebrew-gqrx) and [homebrew-hackrf](https://github.com/robotastic/homebrew-hackrf), but each looked to be more involved than I had hoped with regards to their prereqs.  Homebrew Cask is a way to install binaries through homebrew, and the one I was referring to is actually returned when running `brew search gqrx`

```sh
$ brew search gqrx
Caskroom/cask/gqrx
```
If you know of a better homebrew tap to use for gqrx, please let me know, otherwise I'll look to install it via one of those I listed (or install macports)