gqrx-sdr/gqrx #224: Frequency going in the wrong direction

joushx, 2015-02-01:
When changing the frequency with the freqCtr component, the frequency moves into the wrong direction when using digits after the dot.

The initial state:

[image]

Going up by 0.1 MHz:

[image]

It goes down when it should go up and vice versa.

csete, 2015-02-01:
Ehm, you say "going up by 0.1 MHz" and what I see on the screenshot is that it changes from 95.6 to 95.7 MHz. I'm not sure I understand what you think is wrong here? Perhaps you can elaborate.

joushx, 2015-02-01:
If I increase the frequency the FFT diagram is moving to the right so I can see frequencies to the left of the previous view. Notice the "peak" at the right: I want to go to a higher frequency, but it moves to the right. It was at ~94.65 MHz and is now at ~94.7 MHz. If it would have moved to the left it still would be at ~94.65 MHz.

csete, 2015-02-01:
Sounds like you have the I and Q channels reversed for some reason but it's hard to tell why without knowing anything about your setup.

csete, 2015-02-01:
By the way, try to reproduce this with a real signal, not spurs. Spurs can occur at baseband and then move randomly regardless of whether you tune up or down.

joushx, 2015-02-01:
Oh, I accidentaly checked the "Swap I/Q" checkbox. This resolves the issue. Thank you for your time and your great project!

The said peak was a real radio station by the way.

closed 2015-02-01 by joushx