gqrx-sdr/gqrx #227: GQRX crashes right after starting DSP.

JEvans48, 2015-03-14:
Hi, I've been looking to see if this problem has already been asked and solved elsewhere, and haven't found it yet, my apologies if I've missed it. I did run across the thread regarding libusb and made sure to update to libusb-1.0.19, but the problem persists. rtl_test and rtl_fm both work. I've pasted in the output from running gqrx in debug mode below, hopefully this is just some some dumb configuration mistake on my part, but either way, I'd really appreciate a second set of eyes. 

If I've left out any useful information please let me know and I'll add it in. Thank you.

 ./gqrx
linux; GNU C++ version 4.6.3; Boost_104601; UHD_003.008.002-101-g33edac34

Controlport disabled 
No user supplied config file. Using "default.conf" 
gr-osmosdr v0.1.4-16-g61184a19 (0.1.5git) gnuradio 3.7.6.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using Volk machine: sse4_1_64_orc
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
New filter offset: 0 Hz 
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
BookmarksFile is /home/jevans/.config/gqrx/bookmarks.csv
Loading configuration from: "default.conf" 
Configuration file: "/home/jevans/.config/gqrx/default.conf" 
gr-osmosdr v0.1.4-16-g61184a19 (0.1.5git) gnuradio 3.7.6.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
New antenna selected: "RX" 
******************** 
Gain name: "LNA" 
      min: 0 
      max: 49.6 
     step: 0.5 
******************** 
New audio output device:
   old: 
   new: alsa_output.pci-0000_00_1b.0.analog-stereo
Exact sample rate is: 250000.000414 Hz
IQ DCR samp_rate: 250000
IQ DCR alpha: 3.99998e-06
Changing NB_RX quad rate: 96000 -> 250000
Requested sample rate: 250000 
Actual sample rate   : "250000.000000" 
Requested bandwidth: 0 Hz 
Actual bandwidth   : 0 Hz 
setFreqCorr : 0 ppm 
New LNB LO: 0 Hz 
"New frequnecy range: 24 - 1766 MHz (step is 0 Hz but we use 1 Hz)." 
"New frequnecy range: 24 - 1766 MHz (step is 0 Hz but we use 1 Hz)." 
Filter preset for mode 0 LO: 0 HI: 0 
setFftRate to "15 fps" 
New FFT rate: 15 Hz 
setFftSize to "4096" 
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
New FFT rate: 15 Hz 
New FFT rate: 15 Hz 
Force RX reconf (jerky dongle workarond)... 
rtlsdr_read_async returned with -5
Filter preset for mode 0 LO: 0 HI: 0 
gqrx: pthread_mutex_lock.c:62: __pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.
Aborted

csete, 2015-03-14:
> rtlsdr_read_async returned with -5

Well, even with an updated libusb it seems to be a driver / libusb issue. The fact that rtl-sdr tools work suggests that you may have two sets of tools and / or libraries installed.

JEvans48, 2015-03-14:
Hi, thank you for the feedback, I really appreciate it. That's pretty likely, given the number of tools I had to install before I realized checkinstall was a thing. I've mostly been following tutorials and instructions so far, so I'm kinda stalled on how to proceed. Would you happen to have any suggestions on how to find redundant or conflicting libraries? 

I've got a list of all the software I've installed so far, if that'll help:

Libudev-1.0.19
osmocom.org/rtl-sdr
GNURadio
gr-baz
gr_osmosdr
and then GQRX

I know this is kinda outside troubleshooting for gqrx so I can certainly close the issue and ask around elsewhere.

Thanks again

csete, 2015-03-14:
Looks right except that you wrote libudev-1.0.19 and not libusb-1.0.19. Is that a typo?

JEvans48, 2015-03-14:
Right, sorry. Yes, I meant libusb-1.0.19. I had to install libudev to get libusb to install, that wouldn't have caused a problem would it? Thanks for your help.

JEvans48, 2015-03-15:
Thank you for your help, it's working now. I'd like to be able to leave some guidance in case anyone else stumbles over this, but I basically just ham-handedly broke and reinstalled the tools listed above until it started working. Not my proudest success, but I'll take it.

closed 2015-03-15 by JEvans48