gqrx-sdr/gqrx #228: tried to fix a bug but

paigeadele, 2015-03-16:
Hi, 
There's an issue where gqrx crashes when I try to add the direct_samp=2 parameter to the device string: 

```
gr-osmosdr v0.1.4-16-g61184a19 (0.1.5git) gnuradio v3.7.6.1-126-g892629a3
built-in source types: file rtl rtl_tcp hackrf 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
gr-osmosdr v0.1.4-16-g61184a19 (0.1.5git) gnuradio v3.7.6.1-126-g892629a3
built-in source types: file rtl rtl_tcp hackrf 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::runtime_error'
  what():  In hierarchical block source_impl, output 1 is not connected internally
[1]    22333 abort      gqrx
~  ᐅ 
```

So not being really all that good with C++ I decided to try to add a handler and succeeded in getting it to build (finally) but I can't get it to link: 

```
main.o: In function `main':
main.cpp:(.text.startup+0x19d): undefined reference to `vtable for QCustomApplication'
main.cpp:(.text.startup+0x7ef): undefined reference to `QCustomApplication::~QCustomApplication()'
main.cpp:(.text.startup+0x10f7): undefined reference to `QCustomApplication::~QCustomApplication()'
collect2: error: ld returned 1 exit status
Makefile:246: recipe for target 'gqrx' failed
make: *** [gqrx] Error 1

```

I tried reading about this problem and trying a few things but I can't figure it out. I'd really like to add some other features to gqrx as well but I just don't see it happening if I can't get past the science of C++ and QT programming. I would appreciate it if someone could help me figure this out.

Here's my branch in my fork if you'd like to try to help: 

https://github.com/paigeadele/gqrx/tree/paige

paigeadele, 2015-03-16:
Also just FYI I noticed it does have some information in the error : "In hierarchical block source_impl, output 1 is not connected internally" and I think I repro'd that error with /usr/bin/rtl_sdr. 

http://www.reddit.com/r/RTLSDR/comments/2v8b34/hf_using_chinese_full_band_uv_hf_rtlsdr_usb_tuner/

this is a link to the radio I purchased http://www.ebay.com/itm/100KHz-1-7GHz-Full-Band-U-V-HF-RTL-SDR-USB-Tuner-Receiver-R820T-8232-Radio-US-/191488135592?ssPageName=ADME:X:AAQ:US:1123

I'd like to get this figured out, whether its a driver issue or the hardware just isn't actually what it's advertised as before I leave feedback for this guy. It's better than nothing but its still going to piss me off if its not as advertised.

paigeadele, 2015-03-16:
I tried asking a reddit group where I got the information about how to use this feature, http://www.reddit.com/r/RTLSDR/comments/2v8b34/hf_using_chinese_full_band_uv_hf_rtlsdr_usb_tuner/cpg6l5f

paigeadele, 2015-03-16:
Ah, after reading a bit deeper into the reddit thread I see that using a comma to separate the device string fixes the issue: 

rtl=0,direct_samp=2

starts fine now, not at all what the error message seemed to indicate

csete, 2015-04-03:
So, is there a bug her or can we close this?

closed 2015-04-11 by csete