gqrx-sdr/gqrx #234: Modifying bookmark tags and hitting Cancel removes all tags from bookmark

khaytsus, 2015-03-28:
If you go to modify a bookmark Tags, but hit Cancel, it currently removes all tags from the bookmark.

Using GQRX as of revision f3bb0ab080197b0738dc5713c8222c168f2e1850

csete, 2015-05-18:
Hi @khaytsus  
Could you confirm that the issue is fixed now?

khaytsus, 2015-05-25:
I must be missing something, I have no way to modify an existing Bookmark now?  If I right-click an existing bookmark, I get the options to Delete Bookmark or Add Bookmark.

khaytsus, 2015-05-25:
Okay, I figured out double-clicking the Tag will let me modify the tag.  Others you edit inline.  Working fine now, although a dialog for modifying the whole Bookmark would be nice vs inline.

khaytsus, 2015-05-25:
Tested on 8ffff0c for reference

closed 2015-05-25 by khaytsus