gqrx-sdr/gqrx #236: Bookmark labels in FFT draw over each other in dense channel bookmarks

khaytsus, 2015-03-28:
When there are many bookmarks in one spot in the spectrum, GQRX puts the labels over-top of each other in the FFT area.  It appears to currently only offset by 3 or so vertical levels, so if you have many bookmarks in one place the labels are drawn over top of each other.

It does not appear to matter how large the FFT vs Waterfall is, although I could see that there would be a limitation that the labels would not draw outside of the FFT area.

Screenshot here:  https://dl.dropboxusercontent.com/u/334233/gqrx-bookmark-labels.png

Using GQRX as of revision f3bb0ab080197b0738dc5713c8222c168f2e1850

kolbma, 2020-03-11:
Can you please explain again or provide an accessible image?!

kolbma, 2020-03-12:
[image]
Looks ok I think?!

khaytsus, 2020-03-12:
Well, it is almost 5 years old, I guess I cleaned out my Dropbox ;)  I too do not really recall what this originally looked like, but if you have FRS in your bookmarks, go where those are and zoom out.  This may have been worse 5 years ago, not sure.  https://i.imgur.com/0lALWi4.png

This is FRS/GMRS plus some commercial repeaters etc, standard bookmarks I have, nothing contrived here.  Zoomed out it does look kind of bad, but zoomed in a bit it's okay.  Again, I don't recall the original so it might have been worse.  I don't think as it is this is a big deal.

PR #776 (2020-04-16): Fix for issue #236
dynamic level number based on viewable height
continue with higher frequency if there is no more space for bookmarks
do not overwrite existing markers
precalculate some const reused values in loop

fixes #236
closed 2020-04-16 by csete