gqrx-sdr/gqrx #240: ASFK1200 Decoder no longer seems to work

khaytsus, 2015-04-09:
I am currently using git revision db5ad0e and since moving from the release version of GQRX I've found that AFSK1200 no longer seems to work for decoding APRS packets.  I am able to pipe to direwolf, but it's nice to have APRS data directly in GQRX for quick looks.  I can pipe to direwolf for better data/collection if I need that but I typically personally don't.

I don't know what check-in this might have stopped working, and I've occasionally had trouble with it working in the past but it usually does.  But the last few times I've tried it it's decoded nothing even though there's periodic traffic and direwolf was able to decode many packets at the same time.

csete, 2015-04-09:
I think this decoder will be eliminated very soon. At some point we will add an interface to wrap external applications into mouse clicks but I don't when I will have the time.

michelinok, 2015-04-10:
Don't delete it please.....

csete, 2015-04-11:
I will leave in until there is an alternative, but I will not spend time on fixing it so what's the issue?

michelinok, 2015-04-11:
No issue.... Did not understand that you will leave it there UNTIL there is an alternative

pointhi, 2015-04-29:
I have also compiled gqrx with the git revision 0bbfd3832943739cbe4c3ee2642e4347c4b00753, and for me it is working (I received an arps signal, and there was an correct looking output in the AFSK1200 decoder)

closed 2017-08-19 by csete