gqrx-sdr/gqrx #242: SSB demodulators of gqrx fails to produce well-enough quality audio for fldigi to decode digimodes

enyone, 2015-05-06:
Example video https://www.youtube.com/watch?v=6N-eit1EmY8

This applies nearly all digimodes (mostly bpsk).

vielmetti, 2015-05-06:
I was able to get something to work using the WebSDR + fldigi combo, but not yet with gqrx. See

http://vielmetti.typepad.com/vacuum/2014/02/decoding-radio-digital-modes-without-a-radio-using-websdr-and-fldigi.html

csete, 2015-05-06:
It's probably an issue with bad resampling somewhere but without knowing anything about the setup it's practically impossible to do anything about it. The video is nice but useless for debugging. A detailed description of the setup would be more useful.

vielmetti, 2015-05-06:
For reference here's a second description by OZ9AEC of the connection between a websdr and fldigi:

http://www.oz9aec.net/index.php/component/content/article/63-sdr/290-fun-with-websdr-and-fldigi

In addition this tutorial by Jeff @heliosj228 has some commentary about using a named pipe as your messaging path between components:

https://heliosj.net/main/using_fldigi_to_listen_from_remote_sites

enyone, 2015-05-08:
```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Lifeview LV5TDeluxe
Found Fitipower FC0013 tuner
```
```
Linux rnetl34 3.13.0-37-generic #64-Ubuntu SMP Mon Sep 22 21:28:38 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
```
```
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=17.1
DISTRIB_CODENAME=rebecca
DISTRIB_DESCRIPTION="Linux Mint 17.1 Rebecca"
```
```
model name	: Intel(R) Core(TM) i7-3537U CPU @ 2.00GHz
```
```
linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown
Gqrx software defined radio receiver 
```
```
fldigi 3.22.04
```
```
pulseaudio 4.0
```

csete, 2015-05-08:
> Found 1 device(s):
>  0:  Realtek, RTL2838UHIDIR, SN: 00000001
> Using device 0: Lifeview LV5TDeluxe
> Found Fitipower FC0013 tuner

So, I think there you have the reason. I think rtlsdr devices are pretty bad for digital modes. In particular narrow band modes such as bpsk require good frequency accuracy and these cheap dongles have very poor oscillator.

I think you would have much better success if you tried with better hardware. I you google "gqrx fldigi" you will see many videos showing people doing what you are trying to do but using different hardware.

Various workarounds could be implemented in gqrx but I don't have time at the moment.

PS: I changed the title since it has nothing to do with the SSB demodulator in gqrx

hexameron, 2015-05-15:
This is a bug in fldigi. By default it resamples (using libsamplerate) from the "native" samplerate in your pulseaudio config (default 44100), even though pulseaudio  ignores that and GQRX uses 48000.

You can set the input sample rate in fldigi from "native" to 48000 , or change the rate in the pulseaudio  config from 44100 to 48000.  Which is a pain, but nothing that to do with GQRX.

csete, 2015-05-18:
@hexameron Thanks a lot for this info. Very informative!

hexameron, 2015-05-19:
The change you need to make to Pulseaudio is given here: http://ubuntuforums.org/showthread.php?t=989456 , comment 3.

The Fldigi developers prefer Portaudio, so you could just install that. In Pulse mode Fldigi only resamples for the ppm correction, so the bad resampling happens between libpulse and the pulse server.

enyone, 2015-07-20:
thanks and sorry about blaming gqrx

closed 2015-07-20 by enyone