gqrx-sdr/gqrx #245: low FFT rate

cruvolo, 2015-06-14:
Hi. Using recent gqrx I'm getting low FFT rates, so it seems like the FFT only runs every second or so, despite having 30 fps selected.  Increasing the value to 60 fps still has FFTs running every second.  Seen on two different machines.  One running debian wheezy+backports, the other running debian jessie.

This was seen with gqrx 2.3.1 and a current github clone.  

Device info from lsusb -v:

```
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x2838 RTL2838 DVB-T
  bcdDevice            1.00
  iManufacturer           1 Realtek
  iProduct                2 RTL2838UHIDIR
```

Using ```libgnuradio-fft``` version 3.7.5.

See the waterfall in this screenshot.  This low FFT rate applies to the spectrum scope as well.

[image]

csete, 2015-06-14:
What happens is that data is only delivered to gqrx only once per second. It could be an issue with the hardware, the driver or gnuradio. There is not much I can do in gqrx.

csete, 2015-06-14:
Sometimes it helps to remove and reinsert the device.

cruvolo, 2015-06-14:
Hi, and thanks for the response, but it looks to me like an issue in gqrx (or a dependency).  Here is why.  After leaving it running for a long while (4+ hours), I get more frequent FFTs.  Then, if I change the gqrx receiver mode, it resets to infrequent FFTs!  See this screenshot, where I change the mode from USB to AM:

[image]

You can clearly see the delineation half way down the waterfall where I switched modes and the FFTs reset from more frequent to infrequent.

Also, if the SDR was delivering data only once per second, I would expect to hear distorted audio in the demodulation.  The demodulation sounds fine.

csete, 2015-06-14:
So, if it is a gqrx issue, how come it only happens with rtl-sdr dongles? Gqrx does the same processing regardless of which hardware you use.

What you experience can be explained as follows.

The main DSP processing chain processes all samples regardless of how they come in. So even if the driver delivers a big chunk once every second (or at whatever frequency) all samples will be processed and sent to the audio hardware. The audio hardware will then play it back at it's own rate and it may sound as smooth playback, but you will experience a delay between actions and audio.

The FFT runs in it's own thread. If you set it to refresh at 20 FPS it will check for new data 20 times per second. But if it only gets data once per second, it will have no new data to process 19 times per second.

csete, 2015-06-14:
By the way, I noticed you use FFT size 32k. It means that each FFT needs 32k samples, so  be sure that you use a high enough sample rate. And please, don't use anything less than 1 Msps on rt dongles.

chrisnorthcott, 2015-06-14:
@cruvolo, what hardware are you using?

This is a typical symptom of insufficient USB bandwidth and/or CPU power. Dial down the FFT size to, say, 1024 and FPS to, say, 10 and see if you can reproduce the problem.

I can reproduce this on a Thinkpad T41, but that thing is over a decade old, and does not have anywhere near the CPU power to be able to run gqrx usably.

Have you tried doing a flowgraph (samplerate 1024000, osmocom/RTL-SDR source -> throttle -> waterfall sink) in gnuradio-companion?

If you don't get the same problems in GR with the same settings then yes, it is quite feasible to assume a problem in GQRX, but otherwise the first suspect is your configuration.

johnlaur, 2015-09-18:
I have not looked much into the sources of GQRX, but I was directed to this thread to ask about how gqrx  is feeding samples to the FFT because it seems that the method may not be ideal. Is gqrx dividing the incoming samples into blocks based on the FFT size and then performing FFT on each block? If so this is not a good idea. FFT should be overlapped and performed as many times as required to achieve smooth display. 

Let's say our samples arrive and we call each one by letter:

abcdefghijkl

If we cut them into 4 sample FFT bins as such:

abcd efgh ijkl

we willl then only be able to update the FFT 3 times during that period. If we insted segment the samples as:

abcd
  cdef
    efgh
    ghij
      ijkl

We can update the FFT 6 times per period.

The latter is how it should be being done. Since each sample group needs to be windowed, strong frequency compoents contained in the a,d,h,i,l sampls in the first group will not appear in the FFT.

So, despite samplerate, so long as there is sufficient CPU, the FFT should be able to operate at essentially any desired framerate. "Realtime" FFT such as performed by gr-fosphor and various highend spectrum analyzers run a large number of FFT per frame and display the data as a power gradient.

csete, 2015-09-18:
Original posater did not respond about sample rate. So, under the assumption that he used a "low" sample rate (250k or 300k), the solution is to reduce the buffer size in the rtlsdr driver, see
https://github.com/csete/gqrx/issues/275#issuecomment-139662877

csete, 2015-09-18:
@johnlaur What you dsescribe is already implemented in gqrx. But what do you do if the samples arrive in much larger chunks than your FFT size? Replay them slowly adding even more latency? Hardly optimal... I think reducing the buffer size forcing smaller transfers is a much better way to solve this issue, which only happens with rtl dongles at low rates.

johnlaur, 2015-09-18:
Why not? This is already what you have to do with audio to avoid dropouts so surely the latency is already there. There is no specific need for the FFT to run ahead of the audio. If the audio can playback smoothly there is enough data to run the FFT at any rate you want synchronous to the audio.

But yes running a smaller usb bulk transfer buffer is going to decrease latency if running an RTLSDR at a low samplerate. That may mitigate or work around this problem, but I am not sure it is really a perfect solution. Not a complaint by any means -- thanks for your consideration!

csete, 2015-11-16:
I don't think there is anything to do here, but feel free to correct me if I'm wrong.

closed 2015-11-16 by csete