gqrx-sdr/gqrx #247: Receiver Options Channel Filter Offset Does Not Properly Display Offset Frequency

gqrx, 2015-06-18:
Using latest git copy. Also noticed on latest stable Arch Linux release.

If the channel filter offset is set to any value between -0.001 kHz and -0.999 kHz, the display under "Receiver Options" neglects to include the negative sign. This can be seen clearly in the two images below. The first is set to 100 MHz with a channel filter offset of 0.000 kHz. Then, this channel filter offset value is set to -0.100 kHz. Notice that the frequency display on the left clearly registered that an offset of -0.100 kHz was applied.

[image]
[image]

So, CFreqCtrl::setFrequency(freq) works. However, the way the entire gqrx/qtgui/freqctrl.cpp file is structured, the only way negative numbers are shown in the channel filter offset display on the right is if the leading number is negative. In other words, -1.000 kHz will display properly, but -0.010 kHz will appear as 0.010 kHz.

I'm quite new to SDR, gqrx, and I don't feel comfortable restructuring all that code for such a minor cosmetic bug. I don't know that it is even possible to easily handle. If you forcibly painted a "-" right before the 0, if a user clicked the "-" or used a mouse wheel to scroll up or down or used the up or down arrow keys, the program could then be forced to erase the "-" sign and repaint it with a number.

I'm curious to hear what the developers' take is on this.

Also, thank you very much for this wonderful software!

csete, 2015-06-18:
Thanks for the report.

I am aware of the issue but it never really bothered me enough to do something about it. But lets keep the bug until we fix it.

argilo, 2020-11-28:
It appears this was fixed in 86439028f6b1593d6247a0e9fbf4bc5bfee656d3.

closed 2020-11-28 by argilo