gqrx-sdr/gqrx #248: previous frequency not remebered on relaunch of GQRX

radonclers, 2015-06-21:
I'm using the newest Macports version gqrx 20150615 with an an RTL 2832 device and upconverter for listening to VLF, shortwave and broadcast bands. The RTL device has a range of 24 to 1766 MHZ.  When I enter a frequency less than the minimum (24mHz) and quit, on relaunch the frequency displayed is 871MHz which is exactly half way in the hardware frequency range.  The previous version of gqrx worked fine.  It looks like a new calculation is taking place at startup that blocks frequency values less than the detected hardware minimum from being set and enters a value exactly half way between the minimum and maximum frequencies.  The preferences file is loaded with the correct frequency (less than 24mHz).  Can you suggest a way to fix this.

csete, 2015-06-21:
I will look into it. I did make a change that could be causing this.

Just to be sure, you are using the "No limits" flag, right? I guess otherwise you couldn't set a frequency out of range at all.

csete, 2015-06-21:
Ok, I fixed the issue. It will be available in the next macports update.

closed 2015-06-21 by csete

radonclers, 2015-06-21:
> <reply+00c61c0f2ce838cf31853a937b13917331eaae1093620dc292cf00000001119e561d9
> 2a169ce055af21c@reply.github.com>
> (#248)
>
> Closed #248 <https://github.com/csete/gqrx/issues/248> .
>
> 
> <https://github.com/csete/gqrx/issues/248#event-336122206> .

Thanks for your quick reply Alexandru.

Your software is brilliant !

L
---------------------------------------------------------------------------
Leo Raudonikis
Raudonikis Database Services
ABN 52 079 150 780
18 Flora Ave.
Mt Colah
NSW 2079 Australia

(02) 8005 2246 (VoIP phone)
               (please note voice messages are forwarded to email)

Mobile 0418 129 897

email   :    l.raudonikis@optusnet.com.au
Web     :    http://members.optuszoo.com.au/l.raudonikis/

---------------------------------------------------------------------------

radonclers, 2020-02-24:
Hi Alex,

 

I last contacted you in June 2015. I am currently using GQRX 2.11.5 on a Mac Mini under Mac OS X 10.13.6.  I works very well with an RTL2832 SDR (R820T2 Tuner ) and 50mHz upconverter I built. There don’t seem to be any serious OS related problems.

 

I am now planning purchase of an Airspy HF+ discovery SDR.   I am under the impression GQRX will work with the Airspy HF+ Discovery but would appreciate confirmation or feedback from you.  I am unsure of whether drivers for the Airspy HF+ are present in the Mac version of GQRX because an ‘Airspy’ option does not appear in the hardware configuration dropdown menu. (see below). I noted that the ‘RTL’ option doesn’t appear in the menu unless the dongle is plugged into the USB port before the launch of GQRX software.

 

 

Q1 – Will the Airspy HF+ Discovery work with GQRX 2.11.5 on the Mac (OS X 10.13.6) ?

 

The Airspy HF+ discovery is specified for 0.5kHz to 31 MHz and 60 – 260mHz. So I expect 31 to 60 MHz is a ‘dead band’.

Q2 -  If I were using an Airspy HF+ when I tune GQRX beyond 31MHz does GQRX continue to increase the frequency  32 ... 33 ... 34 ... Mhz  etc or does it ‘jump’ to 60mHz because you have built ‘smarts’ into the GQRX Airspy HF+ driver ?.  

 

Your feedback would be much appreciated !

 

 

best regards

 

Leo

 

 

________________________________________________________________________

 

Leo Raudonikis

Raudonikis Database Services Pty Ltd

ABN 52 079 150 780
18 Flora Ave.
Mt Colah
NSW 2079 Australia

(02) 8005 2246 (VoIP phone) 
               (please note voice messages are forwarded to email)

Mobile 0418 129 897

email   :    l.raudonikis@optusnet.com.au
Web     :    http://members.optuszoo.com.au/l.raudonikis/

 

________________________________________________________________________