gqrx-sdr/gqrx #249: bias-T support for Airspy?

dram-dave, 2015-06-26:
Could you please add Airspy Bias T support to your list for future features, similar to Bias T support for the Funcube Dongle. Thanks!

csete, 2015-06-26:
The interface to enable/disable bias-t is not yet available through gr-osmosdr so it will have to wait. For that reason, it is not available for the Funcube Dongle either.

csete, 2015-06-29:
As it turns out, you can use bias=1 in the device string to enable the bias T.

dram-dave, 2015-06-29:
For us Newbies, could you please describe where/how to set the device string bias=1.  Thanks!

csete, 2015-06-29:
In the I/O configurator dialog there is a field called "Device string". It probably says something like:

<pre>airspy=0</pre>

Change it to:

<pre>airspy=0,bias=1</pre>

(no spaces)

csete, 2015-11-16:
No response for a long time. Assuming current functionality is adequate, but feel free to comment if not.

closed 2015-11-16 by csete

tkeifer, 2016-03-09:
Having issues getting this working over here... 

OS X 10.9.5 with latest master branch compiled from src, gr-osmosdr is v0.1.4

Error in console is:
```
Failed to start RX streaming (-1000)
```

effeerre75, 2017-06-29:
Hi,
with "airspy=0,bias=1" dosen't work at least on OSX 10.12.5 GQRX 2.6, AirSpy Mini.
It work fine with this device string "airspy,bias=1"
In the gr-osmosdr it's also specified to use "airspy" without any value

I've just tried also "Packed USB transmission" (pack) and it dosen't work but this is another story...

csylvain, 2019-07-27:
had to do some deep internet searching to find this. i was mystified for some time why i wasn't seeing signals with my Airspy R2 + Spyverter combo. the combo was working with various other software. explored gqrx looking for a Bias-T check box or similar. 

could the following be an entry in the Documents section of gqrx.dk ?

Airspy: Enabling the Bias-T to power a Spyverter is accomplished by adding ",bias=1" on the end of "Device string" in "Configure I/O devices".

csete, 2019-07-27:
done