gqrx-sdr/gqrx #250: GQRX raw recorded format is Undocumented (or difficult to find)

SterlingPeet, 2015-06-30:
I would like to process recordings in my own gnuradio flow graphs, but the original recording was made in gqrx.  I can determine the hardware frequency and recording time from the name, and I know that it must contain I/Q samples.

When I try to read the file using gnuradio's I/Q file source block, the output seems to be garbage, but gqrx can play back the file just fine (so it must not be corrupted).

I would like to know the file format, so I can convert it to something that can be processed in a gnuradio flow graph.

I recorded the original file using a NooElec DVB+ 820T RTLSDR tuner, and I am running gqrx on OSX 10.10.3 using macports to install gnuradio and gqrx.

csete, 2015-06-30:
The recording is just a dump to a file using the gnuradio file sink block. Use the file source block with complex output, followed by a throttle block using the same sample rate that was used to record the file (also in the file name).

SterlingPeet, 2015-06-30:
Dang, it looks like I had sample rate mismatch in one of my blocks.  Everything looks good now, thanks!

closed 2015-06-30 by SterlingPeet