gqrx-sdr/gqrx #254: GQRX wont compile on Ubuntu 14.04 (LinuxMint) when boost requires -mt flag on pybombs

JannF, 2015-07-04:
As the title says, GQRX won't compile on Linux Mint (Ubuntu based) because it's recipe needs the boost library with -mt flag.
Removing -mt flag from the recipe lets me at least compile.

csete, 2015-07-04:
Thanks, but since the recipe is in pybombs and not in gqrx you should report it there.

ckuethe, 2015-07-18:
I'm working on fixing that recipe... it's been annoying me for a year.

Those bums at Boost still haven't addressed the issue that's been open for over 8 years: https://svn.boost.org/trac/boost/ticket/1094

ckuethe, 2015-07-18:
https://github.com/gnuradio/pybombs/issues/171

mat650, 2015-07-22:
with a fresh pybombs fetch
With mkdir build
cd build
qmake ../
the problem is fixed but Segfault when i 'm trying to start gqrx

ckuethe, 2015-08-01:
Fixed by gnuradio/recipes#38

closed 2015-08-23 by JannF