gqrx-sdr/gqrx #255: Squelch causes delay between waterfall and audio

ozonisator, 2015-07-06:
Hi,
I just recently compiled gnuradio 3.7.7.1 and gqrx v2.3.2-110-g222b from source on ubuntu 14.04 x86 64bit. gqrx works fine, but I noticed a delay between the signal on the waterfall and the demodulated audio while using squelch. The delay gets worse the longer the signal level is below the squelch level. I could repeat this problem several times. I set the squelch level above the noise and let it sit for 5 minutes. I then switch between two signals that are above squelch level, and the demodulated audio lags behind more than a second. This makes capturing short transmissions really difficult because you see the signal on the waterfall, click on it, but you miss it because you hear the old signal for a second. I then tried setting a squelch level that is way lower than ambient noise, but that did not help. It takes a full restart of gqrx for the problem to disappear, and the delay won't  show up until I start using squelch again.
I don't know if the squelch function is implemented in gnuradio or gqrx, so please feel free to delete this issue if this is not a gqrx bug.

csete, 2015-07-06:
Can you confirm that you do not get similar latency with squelch off? Also, which hardware device do you use.

ozonisator, 2015-07-06:
There is no latency if squelch is off. Hardware is rtl2832 with r820t tuner.

csete, 2015-07-06:
Ok, then it is most likely the squelch gnuradio block that introduces the latency. I may replace it with a different implementation.

ozonisator, 2015-07-06:
I tried different demodulators, WFM mono and stereo even corrupts the DSP after about four minutes. Switching the demodulator to another mode seems to fix it. Console shows increasing number of zeros.
[image] 
FM and AM produce only high latency, this is also fixed by switching to another demodulator mode. Thanks for looking into this issue.

csete, 2015-07-06:
To be honest, I'm not convinced that you only see one issue here. The last one you describe sounds a lot like what I have seen before, which seems to be a "feature" of the gnuradio-scheduler combined with large USB buffering in the rtl-sdr driver. I haven't seen it with other devices. But it could be the squelch block causing the trouble.

ik8ozv, 2020-06-09:
Please i have the same issue. I'm noticing as if the signal visible on the waterfall is not synchronized with the audio of the signal. Let me explain: I am tuned to the APRS frequency and I notice the signal on the waterfall first and after 1 or 2 seconds I hear the audio.
Also can hear the background noise with a little hiccup.
As hardware device i am using a Funcube Dongle Pro Plus.
Please any advice?
Thanks
73's de Enzo IK8OZV