gqrx-sdr/gqrx #256: Airspy 10 MHz Sample Rate

whnolan, 2015-07-26:
I am currently using the Airspy receiver with firmware release 1.0.0-RC6. The 2.5 MHz sample rate seems to work fine, but the 10 MHz sample rate causes distorted and garbled audio.

The code was compiled today, and shows version 20150621. Is there support for the 10 MHz sample rate, or is this an issue with the Airspy firmware release 1.0.0-rc6? If the firmware is an issue, what is the latest supported firmware version for the Airspy receiver?

Thanks in advance.

csete, 2015-07-26:
I'm guessing your computer is not fast enough for gqrx to run at 10 MHz sample rate with the settings you are using. You may get lucky by disabling some of the CPU hungry functions (e.g. stereo FM, lower FFT rate and size).

It should work fine but it is some time ago I last checked it. Don't know what version 20150621 means though. Gqrx does not use this kind of version number.

whnolan, 2015-07-26:
The computer has 8 gig of memory, running a quad core i7 processor. The
computer successfully runs the Airspy device under a windows virtual
machine at 10 MHz using the SDR# application. That virtual machine is
restricted to 2 cores, and 4 gig of memory. I am running absolutely nothing
else on the Mac except the GQRX application.

   Possibly what I mistook for a version number refers to the Airspy
device. The Title bar of the GQRX application shows "Gqrx 20150621 -
airspy=0". Again, I complied today, so I assume the version is current. The
Airspy firmware 1.0.0-rc6 was released a few days ago, along with a new
library.

   I am looking for a native OS X application of course. My hardware is far
over-rated for this application/device. I would be happy to assist in any
tests you may want to conduct to resolve. Thank you for the quick response.

Best regards,

Bill

Bill Nolan - KØWHN

Senior Broadcast Engineer

bnolan@rfresults.com

www.rfresults.com

(855) RF Results

RF Results, LLC

P.O. Box 1066

316 E. Willow Road, Suite 200

Enid, OK  73702

(855) 737-3785 x700 Toll Free and Fax

(580) 540-3264 x700 Office and Fax

(316) 778-0734 Mobile

Confidentiality Statement

This e-mail message and any attached files are confidential and are
intended solely for the use of the addressee(s) named above. This
communication may contain material protected by attorney-client, work
product, or other privileges. If you are not the intended recipient or
person responsible for delivering this confidential communication to the
intended recipient, you have received this communication in error, and any
review, use, dissemination, forwarding, printing, copying, or other
distribution of this e-mail message and any attached files is strictly
prohibited. RF Results, LLC reserves the right to monitor any communication
that is created, received, or sent on its network. If you have received
this confidential communication in error, please notify the
sender immediately by reply e-mail message and permanently delete the
original message.

To reply to our email administrator directly, send an email to
postmaster@rfresults.com

vielmetti, 2015-07-26:
Bill, you should be able to install airspy and gqrx from source from Macports. See

http://www.macports.org

and then once you have it installed simply do "sudo port install gqrx" and "sudo port install airspy". Macports keeps track of the exact version you need and any patches, and does the build from source. Note that depending on your machine it might take hours to do the compile (if you haven't yet complied the Fortran compiler, e.g.)

csete, 2015-07-26:
Well, it was just a guess based on the info you provided initially. The behavior does suggest CPU load issue somewhere so it would be helpful if you could check the CPU load.

Can you run airspy_rx utility with 10 MHz sample rate and get the expected throughout? IT should be run with FLOAT32_IQ or INT16_IQ sample type.

You should also run volk_profile to let gnuradio find the most optimal settings.

Knowing that you run OSX I can see that the version number is used by macports, so that's fine.

Unfortunately I don't have any suitable mac to test this setup at the moment.

csete, 2016-05-20:
No feedback for a long time.

closed 2016-05-20 by csete