gqrx-sdr/gqrx #258: It would be really nice if pulseaudio was optional

bogdanr, 2015-08-06:
A lot of distros, especially most of the embedded one don't have and don't want pulseadio and it would be great if gqrx could work without it.

csete, 2015-08-06:
Gqrx can be built to use gnuradio-audio instead of pulseaudio. Gnuradio-audio can then be configured to use portaudio, alsa, jack, windows audio, OSX audio or even OSS.

Can you name just a few of those many distributions that do not have pulseaudio? Not embedded ones because gqrx is not written for embedded systems. I think if you try making gqrx work on an embedded device, pulseaudio will be the least of your problems.

bogdanr, 2015-08-07:
For example Slackware, Puppy Linux don't have Pulseaudio. CentOS has it but it doesn't seem to work properly.

How do you build Qgrx with Gnuradio-audio ?

csete, 2015-08-08:
Look in the gqrx.pro file (not the qgrx.pro ;-)

csete, 2015-12-24:
With the latest code from git master, you should now be able to explicitly request gr-audio backend if you build with cmake, see #312

closed 2015-12-24 by csete

bogdanr, 2015-12-27:
Great this went through, but how would I actually build this with cmake?
Does it also work with QT4 because when I attempted it complained that I am missing QT5?

csete, 2015-12-27:
You add -DLINUX_AUDIO_BACKEND=Gr-audio to the cmake command line.

I think the cmake build only works with Qt 5, but I'm not sure.

csete, 2015-12-27:
By the way, even if you use a Linux without Qt 5, you can download a binary from http://qt.io and install it in a custom location where it does not interfere with the system Qt. That's what I do on all my computers.