gqrx-sdr/gqrx #259: [Enhancement] Add Peaberry SDR V2 Support to GQRX

Boggartfly, 2015-08-18:
A lot of people own peaberry SDRs RXTX but there are no full fledged applications for Linux and OSX Mac platforms that can use peaberrys. Windows uses HDSDR which is the only reliable applications that works for me. I would like to see support for this particular piece of hardware on Linux and OSx as well.
Infact I was contemplating writing some software my self from a fork. The original code is written by @AE9RB at https://github.com/AE9RB/peaberry-cw . It does only CW, but I hope more for it.
**Please consider adding this hardware with TX support with gqrx or please atleast guide me to do so on the other fork**

Boggartfly, 2015-08-18:
Also to be clear, I'm a licensed amateur radio operator from India with the callsign VU3TTL.

csete, 2015-08-18:
Gqrx does not interface directly to any hardware but uses the gr-osmosdr library:
http://sdr.osmocom.org/trac/wiki/GrOsmoSDR
Support for any new hardware has to be added here and will then be available in gqrx with minimal changes.

As for TX support, gqrx is a receive only application and can not TX.

Boggartfly, 2015-08-19:
Thanks for the reply, I'm thinking of making my own code. I'll close this issue for now..

closed 2015-08-19 by Boggartfly