gqrx-sdr/gqrx #264: Incorrect bandwidth display in spectrum view

darrenfreeman, 2015-09-05:
I saved a bookmark at "wide" filter bandwidth setting. Then I opened bookmarks.csv and I saw that this corresponded to 200 kHz.

I ran the application again and selected this bookmark. It loaded the bandwidth as "user (200 k)".

This is not really expected, I expected it to restore as "wide", but okay. Then I changed it to "wide" and the bandwidth displayed for this filter actually increased! This is repeatable, I can change between "wide" and "user (200 k)" and the bandwidth for user is actually narrower when shown over the FFT spectrum.

But when I set it to "wide" and attempt to edit the bandwidth graphically, by dragging on the right edge of the filter shaded region, the tool-tip that pops up says the high cut is at 100 kHz. So I'm confident that "wide" really means 200 kHz. So what is going on with "user (200 k)" if it is narrower than 200 kHz? In fact it also shows as a high cut of 100 kHz when I mouse over to it in the spectrum display. So the tool-tip says they are the same, but the shaded region definitely differs.

Are they the same, or not? And why does it import the book-mark as "user (200 k)" if they are the same? I think something is wrong here.

csete, 2015-09-05:
> Are they the same, or not? And why does it import the book-mark as "user (200 k)" if they are the same? I think something is wrong here.

They are the same in the sense that "Wide" just applies a preset corresponding to 200 kHz in wide FM mode. Currently, stored filter settings are applied directly without checking whether it is the same as a preset.

What can I say, it's a work in progress.

darrenfreeman, 2015-09-06:
"What can I say, it's a work in progress."

Well, I really like it so far, and I've been using it for maybe two days. Just trolling around the spectrum, trying to look up what I am seeing. I've added a lot of bookmarks just to say, yes, I know what this is. Many of them I can't demodulate.

There is definitely something funny going on with the filter bandwidth, either in display or in implementation. I was looking at my 36 MHz radio control, which I know is about 10 kHz bandwidth because it's been certified by someone with a good spectrum analyser. But technically it's only certified at 20 kHz, so I set the bandwidth to 20 kHz. Relative to the signal, the filter was clearly drawing as only 10 kHz wide.

I tried dragging the edge of the filter to manually resize it, and when I put it where I thought it should be, that's when it showed "user (20 kHz)".

So there's definitely a factor of two bug in here. I imagine that in one place you divide by two to get the edge frequencies as +- 0.5 BW, and then you do it a second time in another function by mistake.

csete, 2015-09-06:
Sounds strange. I haven't noticed such issues except that the value in the filter width combo is not updated when changing demodulator.

Note that some modes (AM, FM) use symmetrical filter, i.e. ± 0.5 BW, while others (SSB, CW) use sidebands with filter 0 ... BW or -BW ... 0. But the width should be correct.

Also note that 10 kHz bandwidth only means that X% of the signal power is within 10 kHz, but what you see can be narrower or wider depending on many factors such as, signal strength and FFT resolution.

darrenfreeman, 2015-09-06:
I have just found that when I load a bookmark, it sets the filter bandwidth correctly (you can hear this), but it displays the bandwidth for a different entry in the combo box.

And when I change from normal to user(xx), and back again, it only displays normal. You can hear that the filter is really changing, so I think the transition from normal to user(xx) is not updating the displayed width but it is updating the actual width. Possibly from any preset to user(xx) is causing this problem.