gqrx-sdr/gqrx #265: Feature request: save bands as well as bookmarks

darrenfreeman, 2015-09-05:
What I would like to be able to do is save a band, being a range of frequencies.

Right now I have a book-mark for "70 cm band low" and "70 cm band high", which lets me see visually where the 70 cm band is in my country. I would like to instead be able to save a band as a bookmark-like object specifying two frequencies, and instead of drawing it like a pair of bookmarks, it would draw a shaded region between those two frequencies. Maybe you'd only get the name "70 cm band" if you mouse-over the region.

The band could also come with default modulation and filter width. You could right-click on the band and select "load defaults for this band", whenever you saw an interesting signal and had an idle curiosity about it. Probably most useful for commercial radio stations, where the format is well defined. A band could also specify a default channel spacing, for example "snap to 900 kHz grid" would work well in Australia. This could perhaps display as a group of no-name bookmarks through the band. If you select any of these, it loads the default settings for that band and snaps to that channel.

I think that would be pretty useful. I'd tag everything.. mobile phone frequencies, ISM bands, everything. It' the sort of thing that could one day be bundled in the git repository, at least for those bands that are fairly region independent.

rwnobrega, 2015-09-06:
Maybe this is a duplicate of #205?

csete, 2015-09-06:
> Maybe this is a duplicate of #205?

Yes, they are more or less the same. I will leave the details to whoever is going to work on this.