gqrx-sdr/gqrx #268: Tag view not wide enough in create bookmark dialogue

darrenfreeman, 2015-09-05:
When adding a bookmark, if there are tags in use, then the list of tags appears under the proposed name of the bookmark.

For some reason, the tags list is not the full width of the window, it's only maybe a third of it. So long tags are cut off and you have to guess what they are called. There's no way to resize this.

rwnobrega, 2015-09-06:
I believe that 72db4d8a4977fb6bf4cf0dea94c1b8715f7581fb fixes this.  (Incidentally, the commit also replaces some tabs with spaces. I hope this is okay.)

csete, 2015-09-06:
> I believe that 72db4d8 fixes this. (Incidentally, the commit also replaces some tabs with spaces. I hope this is okay.)

Hmm, I only see whitespace changes in commit 72db4d8 ?
Btw. I do prefer spaces instead of tabs.

rwnobrega, 2015-09-06:
Oops, my bad! Fixed now in e606228788882e006712515d66b34fa3939201fa.

commit 3a516184 (2015-09-06): Expand last column of bookmarks dialog tag list (fixes #268)
closed 2015-09-06 by csete