gqrx-sdr/gqrx #271: Feature request: disable audio FFT, also switch to waveform view

darrenfreeman, 2015-09-06:
I am trying to make gqrx run on a little netbook with Atom processor. It can't keep up with wide FM demod, it's just not fast enough. So I was trying to turn off as much as possible, FFT frame rate set to zero, etc.

However it still performs an FFT on the audio output. I would like to be able to turn this off. Does it stop calculating if that window is closed? Can I just disable it without closing that window?

I would also like to be able to see the waveform. If I look at my radio control at 36 MHz, I should be able to see a demodulated square PPM pulse train. But all I can do is listen to it and see an FFT, which is not helpful here. So I would like to be able to toggle between audio FFT, audio waveform, and no audio display. I would also like a mute button for when I want to demodulate non-audio signals.

Initially it would be helpful to just display the waveform unsynchronised, but sometime in the distant future, I would like to be able to set a trigger level, and have it trigger just like an oscilloscope.

csete, 2015-09-06:
Sorry, but gqrx and in particular wide FM require much CPU and wont work on small processors. I think you will win very little by disabling audio FFT. It's not possible to disable audio FFT without closing the window.

darrenfreeman, 2015-09-06:
I managed to get it working just fine. The solution is to drop the input sample rate to just what you need, say 300 kHz. You don't get the nice waterfall showing the two adjacent channels, but you do get to listen to the one that you want without it shuddering.

So I wonder if it's possible to have a friendly button that's right where you can easily find it, maybe below the demod mode setting, that automatically drops the sample rate based on the filter settings that you've chosen? The RTL-SDR dongle is presumably filtering it before downsampling, so might as well let it do more of the work. In fact it would be nice if that meant that you didn't need to filter it a second time in software prior to demodulation.

The friendly button that I mentioned above would toggle between two states, so you can have a nice wide input bandwidth for locating something interesting, then when you want to demodulate it, you would toggle the input device into narrower bandwidth so your Netbook can handle demodulating it. And when done, you could toggle back the earlier settings. The FFT window would update its frequency units accordingly. Maybe there would be a check box that mutes the demodulation on the wider input bandwidth, since stuttering audio is horrible, and it robs CPU cycles from the waterfall, which is what you are interested in when the bandwidth is wide.

I think it can be made to work well on a Netbook as a portable radio scanner.

darrenfreeman, 2015-09-06:
I can see myself contributing a patch one day, as I have worked with Qt and have a background in signal processing. So please don't feel like I'm overloading you with feature requests, they're just place-holders for ideas that I might end up helping with myself some day.