gqrx-sdr/gqrx #273: FFT view gets displayed off-centre after zooming in a long way, then back out

darrenfreeman, 2015-09-06:
Drag the carrier marker off-centre. (Now there should be displayed an offset under receive settings.)

Go to FFT settings, and zoom in a long way, e.g. 50x. (Why is this control here? Why can't I zoom by interacting with the FFT visualisation?)

Now zoom back out to 1x. The FFT is displayed off-centre. On one side, is black. The other side is hidden below other window widgets. The centre is permanently offset, and the application must be restarted to get it back to normal.

csete, 2015-09-06:
> Why is this control here? 

Because that's where the other FFT settings are located.

> Why can't I zoom by interacting with the FFT visualisation?

You can. When the mouse is over the frequency or the dB axis it changes shape. You can then drag the spectrum by click-and-drag or zoom using the mouse wheel.

> Now zoom back out to 1x. The FFT is displayed off-centre. On one side, is black. The other side is hidden below other window widgets. The centre is permanently offset, and the application must be restarted to get it back to normal.

I can't reproduce this so I suspect you either dragged the spectrum off center or maybe pressed the "D" button. In either case, pressing the "R" button will reset all zoom and offset settings.

darrenfreeman, 2015-09-06:
Thanks for clarification of the much easier way to zoom in and out!

"I suspect you either dragged the spectrum off center or maybe pressed the 'D' button."

I don't believe so. Are you sure you're using a large frequency offset in the receiver? Like right over to one side of the spectrum?

darrenfreeman, 2015-09-06:
Maybe I have found a bug after all..

When I scroll-wheel zoom the FFT, the zoom control under FFT settings is not updated, even when it is visible.

darrenfreeman, 2015-09-06:
And if I hit the R button, it is not updated.

argilo, 2020-11-28:
This issue does not seem to be present anymore.

closed 2020-11-28 by argilo