gqrx-sdr/gqrx #274: Feature request: direct entry of receive frequency

darrenfreeman, 2015-09-06:
The part where you flip each digit of the frequency up and down is fun for a while, but there are times when you wish you could type it in with the keyboard.

I was demoing to a couple of people in a cafe today, and I forgot to bring my mouse. I had no way to input mouse-wheel events, and it took me a while to work out that you can make the numbers go up by clicking in the top of the digit, down by clicking in the bottom.

This was one of those times when being able to bring up a widget that accepts numerical keyboard entry would have saved quite a lot of time and frustration. My friend was reading out quite different frequencies to see if we could pick them up, and it took ages to enter them.

Also it would be nice if there was a button to zero the offset frequency. The most common operation on the offset was to zero it after dragging the carrier marker around graphically. Without zeroing the offset, if you load a book-mark, you still have an offset. So you then zero it one digit at a time, then you need to load the book-mark again.

csete, 2015-09-06:
When the frequency entry widget has focus, you can use the arrow keys or enter a frequency directly. A shortcut to give focus to the frequency entry widget might make sense, but I see no reason for a new widget.

darrenfreeman, 2015-09-06:
I see that now, thanks.

Honestly I find that widget quite counter-intuitive. I gather it's a custom one, and I haven't seen anything like it in another application. I'm sure I will get the hang of it but I hope it can be made a little easier to just pick up and use without reading the manual.

benburkhart1, 2015-09-19:
Hello @csete,

What behavior do you expect when someone starts typing when the focus is anywhere other than the frequency control? From what I can tell it does nothing except when focused of course on a edit box. Would you accept a patch that would allow free-typing the frequency when not focused on an editbox or the hardware freq frequency display?

darrenfreeman, 2015-09-19:
I think the issue is that you can hardly tell whether it has focus, and which digit is current.

@benburkhart1 the problem is that you can select any digit in the widget, and when you start typing, it only overwrites from that digit onwards. You might have 150.123456 MHz, and want 150.123765 MHz, which you can get by clicking on the 4 and typing 765. This is nice. The problem is every time I start typing, I'm always doubtful as to which digit I'm going to overwrite.

There would be plenty of other places where direct entry is appropriate, if not now, then in the future. Entry of bandwidths, offsets, squelch levels, etc. Having it always update the frequency would lead to accidental changes.

I propose that the frequency widget gain a proper cursor, perhaps a solid rectangle drawn around it, with a UI setting to either make it blink at 1 Hz, or not. And the digits should have up/down arrows to signify where you need to click to cycle the digits up and down. (Think of a hardware equivalent, often you have thumbwheels and/or arrows.) Perhaps the arrows are only drawn for the digit that has focus, if there is one.

Right now, it's very permissive about changing the digit when you click just about anywhere in the digit. But to facilitate selecting a digit without changing it, the up/down clickable region should be shrunk to just where the arrows are, leaving most of the digit able to be selected without causing a change.

If the widget should lose focus, then the cursor rectangle would go away. You wouldn't be able to start typing or using the arrow keys until you give focus back to the widget, consistent with every other keyboard interactive widget.

Perhaps then the overall application would have a hot-key for entering the frequency, the offset, squelch level, and anything else that can gain keyboard focus. E.g. if I press F and then start typing, it gives focus to the frequency widget and starts updating from the last digit that I edited. I can use the arrows if I want a different digit. Then I hit O and I can start typing an offset. It brings the dockable "receive" window to the front if it is hidden.

csete, 2015-09-19:
@benburkhart1 A patch is welcome, but as @darrenfreeman points out, you don't know if the user wants to enter a kHz, MHz og GHz frequency. So the behavior will be confusing in another way.

I think the way to do it is to enter the frequency from backwards until the user presses enter.

csete, 2015-09-19:
@darrenfreeman I would kindly request that you stop expanding the scope of existing issues and stick to the original topic. I appreciate all the ideas you have and spend time to write them down, but adding new requests to existing items only makes them confusing and less probable of ever getting done. 

So, lets stick to the original topic: Improving the frequency entry widget.

If you want shortcuts for everything else in gqrx, please open a new feature request, although I wonder if not Linrad might be better suited for that purpose.

darrenfreeman, 2015-09-19:
@csete is there a place where the future design of gqrx is discussed? I only raise other ideas because if you fix an issue once, and then it's not compatible with where the overall application ends up, then you end up fixing it twice.

csete, 2017-03-19:
Commit 36f94578d5abe2cbb626c83b2c122895a5a865 adds a widget for manually entering frequency.

closed 2017-03-19 by csete