gqrx-sdr/gqrx #280: UI corruption by changing the sample rate after the radio has been run

darrenfreeman, 2015-09-12:
Try it. Run the radio. Stop the radio. Open device settings, change sample rate from 2.5 MHz to 2.6 MHz, hit okay.

The FFT resizes, the dock widgets on the right hand side resize, but everything is corrupted.

Resizing the window seems to always fix it.

darrenfreeman, 2015-09-12:
I think that when it calculates the size of the widgets after the sample rate change, it is using the size of the window as if it's not maximised, but I have it maximised.

So when I double-click on the title bar (in KDE) to un-maximise it, then nothing inside the window changes, because it was already that size.

csete, 2015-09-12:
Sounds like an issue with the version of Qt you are running, perhaps in combination with KDE because I can't reproduce it.

darrenfreeman, 2015-09-12:
Can I confirm that you are running the application in its maximised state?

Try changing the shape of the window before doing this. It's as if it's either recalculating based on the size that the window would be if not already maximised by the window manager, OR it's using a hard-coded initial value for the window, which you would get if you ran the application for the first time.

csete, 2015-09-12:
I am aware of issue that the window restores its original size when the device configuration dialog is closed, but I don't know what you mean by "everything is corrupted". Perhaps you can try to explain again or add a screenshot illustrating it.

darrenfreeman, 2015-09-12:
I see the problem. On my system, the window doesn't change size when the device configuration dialogue is closed. It paints as though it has changed size, but the window hasn't changed size because it is maximised. Maybe this is an interaction between Qt and my window manager (Plasma).

Qt thinks the size has changed, but it hasn't. But also you might be able to see this on your system too, if you play with maximising it.

csete, 2017-03-19:
Works for me. Sounds like a desktop issue.

closed 2017-03-19 by csete