gqrx-sdr/gqrx #282: AOS command should open squelch or have ability to set squelch from remote control

khaytsus, 2015-09-23:
When I'm scanning and recording using my gqrx-scan script, I've found that if I have the squelch set too tight in GQRX I will sometimes record empty or choppy audio.

For example, my scanning script has detected a signal and I've set -65 is the threshold for it to send AOS to GQRX, but GQRX is set to -60 squelch.  In this case, it doesn't record any audio, only silence.

This happens often as different bands have very different squelch levels...  70cm I typically see about a -65db noise floor but 2m I often see more like -40db.  A workaround is to mute gqrx's audio output and set the squelch so it's always open, but that's not really ideal.

It would be nice if AOS either opened the squelch, or alternatively there was a way to set the squelch via the remote control TCP interface.

csete, 2015-09-23:
I think the right thing to do would be to add squelch level to the interface because I don't like the idea of AOS having a side effect. Can you try to add it? I am busy at the moment and don't know when I will have time to do it

khaytsus, 2015-10-01:
I started on this a bit a few days ago, but haven't done C++ in a few years, and never QT, there's a lot of glue to put in random places..  I've been following the pattern of change frequency.  If I do get something that works I'll fork and issue a pull request.  :)

csete, 2015-10-01:
Sounds great. If you have specific questions or get stuck just ask.

khaytsus, 2016-01-03:
I never got very far with this, I never could get all of the glue tied together.  I followed some very similar functionality but I just made more and more of a mess ;)

khaytsus, 2020-04-16:
Pretty sure #328 fixed this long ago :)

closed 2020-04-16 by khaytsus