gqrx-sdr/gqrx #287: SDR kernel API support

mercuriete, 2015-10-22:
its been more than 1 year that the kernel have a kernel API for sdr devices.

The benefit of using a kernel driver instead a user space driver is that you dont need to be worried
about inserting modules, recompiling or blacklisting modules.

Another benefit is as the API is equal for all devices, you only need to code once and all devices that works with this new api will work fine as well.

I wanted to know if this project have intentions to implement support for SDR kernel API because i just bought an rtl2838 usb dongle and the rtl_sdr kernel module allows to use SDR features and you dont need to unload the module or reboot in order to see DVB-T broadcast.

I think that having support for SDR kernel API will encourage to new people to get into software defined radio because its plug & play and we wont need to install anything.

there are more information in the antii palossari blog (v4l2 kernel programmer)
http://blog.palosaari.fi/2014/01/hacking-with-sdr-api-gain-contols.html
https://github.com/palosaari

Thanks in advance.

csete, 2015-10-22:
Gqrx uses the gr-osmosdr library, which already provides a unified API for a lot of SDR hardware on both Linux, BSD, Mac and Windows. Switching to a Linux only API that only supports a small subset of hardware and even that only partially seems like a very bad idea.

IMHO, the Linux SDR API is nothing but a failed proof of concept that has no practical application. It really blows my mind that it has actually been included in the kernel without any evident collaboration with the SDR community.

closed 2015-10-22 by csete

mercuriete, 2015-10-23:
thank you for your answer.
i finally got working my rtl sdr with osmosdr and works fine.
thank you for this amazing tool.

:)