gqrx-sdr/gqrx #288: Feature request: 2MHz sample rate for "HackRF One"

ly2ss, 2015-11-08:
I was able to enter it manually and it works very good IMHO.

Slightly off-topic, but related
1. at this sample rate would be good to get bigger FFT size. (65536 ? Of course, if math behind it allows it . It's just for better resolution for closely located signals, f.e. ham contests or DX pileups):
https://www.dropbox.com/s/ha6jy12ipzztxxx/fft.png?dl=0

2. at this sample rate there are a bit uncomfortable delays  - first appears the signal on the screen and after noticeable delay (half a second) sound appears in phones. No signs on delay with bigger sample rates. The same delay observable when retuning.

csete, 2015-11-08:
Hi @ly2ss 

The problem with lower sample rates is that the anti-aliasing filters do not provide sufficient attenuation at these bandwidths. So, while it works from a functional point of view, styrong out of band signals will create ghost images in the received spectrum. See more details here:
https://github.com/mossmann/hackrf/wiki/Tips-and-Tricks

I will try adding software decimation like it exists in e.g. SDR# but I can't promise anything.

Concerning the delay, lowering the buffer size in the device string may help, see http://sdr.osmocom.org/trac/wiki/GrOsmoSDR#HackRFSourceSink

csete, 2015-11-09:
I have increased the maximum FFT size to 1M points. Note that the the first time you use a new large FFT size, gqrx may hang for a few seconds because FFTW has to create a new plan. This happens only the first time. I can look into fixing this if it is too annoying.

[image]

ly2ss, 2015-11-10:
Beautiful! Works very very well. On 8 MHz sample rate and 1M FFT size a PC feels a bit sluggish but acceptable (i5@ 2.7GHz). On 2MHz s/r feels really good.
That FFT switching delay is OK IMHO, It's not needed to switch it frequently. No annoyance really.
73! Thank you!

closed 2015-11-10 by ly2ss