gqrx-sdr/gqrx #299: Can't run gqrx on ubuntu

gabika1941, 2015-11-18:
I might sound stupid, but I have a problem. I've compiled the gqrx successfully, but I don't see an executable "gqrx" file in the build directory. 
I've navigated to /build/src, and there was a file called gqrx, but when I've tried to run it, it said "segmentation fault"
What did I do wrong? 
(Ubuntu 15.04, ODROID XU3)

csete, 2015-11-18:
Gabor,

You don't sound stupid but I think you may get more help on the mailing list, since that is where users are: https://groups.google.com/forum/#!forum/gqrx

You will probably need to provide more info about what steps you took and your hardware setup before anyone will be able to help you. Also be sure to check the list of known issues for the hardware device you are using and if possible, test your hardware with a command line utility that comes with the driver.

I haven't tried the odroid myself so I have no idea if it works or not on an odroid. I may try it later, but right now I am busy getting a new release finished.

gabika1941, 2015-11-20:
I want to use an RTL-SDR stick. I've installed gnuradio, and the computer can activate the SDR dongle, it receives signal, rtl_tcp works, too. 
I've seen a video on youtube about gqrx running on an ODROID C1.

csete, 2015-11-20:
Did you install gnuradio packages from the package repository or compiled yourself from source?

I will try to see if I can get it built but I can only test on Odroid C1+ and XU4.

csete, 2015-11-20:
Also, do you see any error messages in the terminal or in the syslog?

csete, 2015-12-20:
No response for weeks.

closed 2015-12-20 by csete