gqrx-sdr/gqrx #302: Airspy bit-packing support?

dram-dave, 2015-11-26:
Airspy's latest firmware release (v1.0.0-rc6) now supports bit-packing ("packing ADC samples to 12bits instead of using 16bits, packing uses 25% less USB bandwidth versus No Packing").  Can this feature somehow be used in conjunction with gqrx?  This might help marginal computers/USB interfaces handle Airspy's 10 Ms/s rate.

csete, 2015-11-26:
Have you tried it?
Since gqrx does not do any hardware interfacing there is not much we can do in gqrx. If bit packing has to be enabled, it has to be done in gr-osmosdr.

I doubt though that any computer not able to transfer the 10Msps rate will be able to run gqrx with that amount of data because I have seen the opposite.

csete, 2015-11-26:
I looked in the libairspy source code and it seem it is disabled by default, so it has to be enabled. Still, the airspy API is not available to gqrx, so it can only be added through gr-osmosdr.

martinzsmith, 2016-05-02:
new argument in the latests  gr-osmosd for the airspy pack=1

csete, 2016-05-02:
I have updated the PPA last night so people using it should have this available now.

csete, 2016-07-05:
I can confirm that adding pack=1 to the device string works, although often gqrx will not be able to reopen the Airspy and require a restart of the application.

closed 2016-07-05 by csete