gqrx-sdr/gqrx #303: Remote control is broken.

kukabu, 2015-11-30:
Hello

I trying e4ddf77b21e9bc39ee840a281f7e92f5332447ea to use remote control but connection immediately closed.
```shell
$ telnet 127.0.0.1 7356
Trying 127.0.0.1...
Connected to 127.0.0.1.
Escape character is '^]'.
Connection closed by foreign host.
```

tomswartz07, 2015-11-30:
I'm also seeing this issue on release version 2.3.2, ArchLinux.

csete, 2015-11-30:
Hi, 

Thanks for reporting this. I didn't realize it wasn't working.

I think it has to do with the transition to IPv6. For instance, my computer no longer has localhost IP address 127.0.0.1 but ::ffff:127.0.0.1, so I had to add this address to the list of allowed hosts.

I have just committed some changes that will print the IP address of the client in the terminal so you can see which address you have to add to the list.

@tomswartz07 I'm not sure ifit will work in 2.3.2 - As far as I recall, IPv6 compatibility has been added to gqrx after v2.3.2 but I am not sure.

Let me know if it works or not.

kukabu, 2015-12-01:
@cset, thanks. Yes, gqrx wants IPv4-mapped IPv6 address.
```
*** Remote connection attempt from ::ffff:127.0.0.1 (not in allowed list)
```

csete, 2015-12-07:
I'm updating the default to be ::ffff:127.0.0.1.

closed 2015-12-07 by csete

tvanden, 2016-08-13:
@tomswartz07 hey, did you figure it out? I am on arch as well. What I had to do was "ping localhost" which showed that my localhost ip was ::1 I had to add that to the allowed list under the remote control settings. Then it worked. Hope this helps you, or someone else.

Sciss, 2017-10-13:
Can you please re-open this issue; it's not fixed. It's broken on the Raspberry Pi:

    *** Remote connection attempt from 127.0.0.1 (not in allowed list)

This is grave as gqrx apparently doesn't remember the whitelist, so adding "127.0.0.1" doesn't persist between program starts. Thanks

csete, 2017-10-13:
@Sciss I can reopen it, but are you sure it is this issue you are having. Gqrx not remembering 127.0.0.1 sounds more like issue #559 but the remote control version.

Sciss, 2017-10-13:
Is there a way to add `tcp_host=127.0.0.1` or something similar to the `.conf` file? which section? I also tried creating the socket as `new java.net.Socket("::ffff:127.0.0.1", 7356)`, but the same message appears; I guess the Pi doesn't run IPv6 (and I can't change it now as the node is integrated with a sound installation). If there is a way to _add_ the settings to the .conf file, that could perhaps fix it. I just don't know what the key would be?

Sciss, 2017-10-13:
Ok, I found a work-around. The settings get read (although not written, it seems). So if I add

    [remote_control}
    allowed_hosts=127.0.0.1

It works for me on the Pi,

csete, 2017-10-13:
Try adding an extra host so you have two, I think it may be saved then. I just saw a bug in the code which looks like if there is only one host listed it will not be written.

asbestomolesto, 2023-03-29:
This issue is still there. I spent almost one hour of tries  before finding that I need to add ::1 to allowed hosts list.

argilo, 2023-03-29:
@asbestomolesto Which version of Gqrx are you using? Version 2.15.8 made improvements to IP address checking (see https://github.com/gqrx-sdr/gqrx/pull/1088 for details).

asbestomolesto, 2023-03-30:
Mine is 2.14.4 - Thank you for the advice, I will try the newer version. It's running on a thinclient so compilation will take a lot of time, but I will try :) 
OS is Devuan GNU/Linux.

commit 6948914d (2015-12-07): Use IPv4-mapped IPv6 localhost as default.

Closes #303.