gqrx-sdr/gqrx #304: RDS decoder doesn't work

kukabu, 2015-12-02:
Hello

RDS decoder doesn't work. May be it's because gqrx doesn't linked or loaded(?) libgnuradio-rds.

csete, 2015-12-02:
Hmmm... seems to work fine for me, although it does require a strong and clean signal.
[image]

kukabu, 2015-12-03:
Hm... then it's may be problem with my system (Fedora 22). I try to investigate an issue.

ps. rds_rx.grc from gr-rds works for me.

csete, 2015-12-03:
If you are building with cmake, try building with qmake and see if it make any difference.

kukabu, 2015-12-04:
I builded with qmake, I rebuilds with cmake. RDS doesn't work.
Are you gqrx loads the libgnuradio-rds?
```
$ lsof -p 3180 | fgrep libgnuradio
gqrx    3180 mike  mem       REG              253,0    271680   463736 /usr/lib64/libgnuradio-uhd-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0     83416   435788 /usr/lib64/libgnuradio-fcdproplus.so.0.0.0
gqrx    3180 mike  mem       REG              253,0     99528   463705 /usr/lib64/libgnuradio-fcd-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0     71064   416735 /usr/lib64/libgnuradio-iqbalance.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    647168   418439 /usr/lib64/libgnuradio-osmosdr-0.1.4git.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    939512   463722 /usr/lib64/libgnuradio-filter-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    183584   463709 /usr/lib64/libgnuradio-fft-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0   1144072   463701 /usr/lib64/libgnuradio-digital-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0   2721152   463697 /usr/lib64/libgnuradio-blocks-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    281888   463695 /usr/lib64/libgnuradio-audio-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    503272   463691 /usr/lib64/libgnuradio-analog-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    333760   463728 /usr/lib64/libgnuradio-pmt-3.7.7.so.0.0.0
gqrx    3180 mike  mem       REG              253,0    975992   463732 /usr/lib64/libgnuradio-runtime-3.7.7.so.0.0.0
```
but you're right, I take a weak signal and not clean signal.

csete, 2015-12-04:
We don't use libgnuradio-rds. Besides, these are libraries linked at compile time and not "plugins" loaded at runtime. If a library is missing you get an error when you try to start gqrx.

kukabu, 2015-12-04:
Oh, gqrx included gr-rds sources and uses it so that gqrx doesn't linked with libgnuradio-rds.

csete, 2015-12-20:
Can we close this? I have no idea what to do here since it works for me as it should.

kukabu, 2015-12-21:
ok, if I find something that will reopen.

closed 2015-12-21 by kukabu

csete, 2015-12-24:
As it turns out FM stereo was broken and it could have had impact on RDS performance. The change that broke FM stereo is now reverted, so you could give it another try.

kukabu, 2015-12-25:
Thanks! It works now.

minexew, 2016-09-06:
Came here just to find out that the checkbox labelled "Disabled" actually means "Enabled".
This is some really, REALLY bad UX.

csete, 2016-09-06:
Hi @minexew 

Thanks for pointing out that we suck at UX. Hope you have time to send a patch that fixes it, otherwise the UX will keep being bad until somebody else has the time to fix it.

Potomac, 2021-04-21:
This bug may exist again in 2021, I didn't succeed to have RDS infos with gqrx 2.14.4, I use archlinux and a hackRF with the last firmware (2021-03).
It can be also due to a weak FM signal but not sure.

kosmas12, 2024-08-12:
I tried this today with my RTL2838-based dongle (I believe it's an Airspy, but can't remember and it's not written on it) and I can't get RDS even though I'm listening to a station very close to my house with -12 to -13 dbfs signal... I'm not sure what might be the cause. Gqrx was just installed from the Linux Mint repos. Everything else works. Hardware AGC is turned on, but I don't think that could be a problem. Other than that, I've changed no receiver settings

argilo, 2024-08-17:
The signal to noise ratio looks quite low. You'll likely need a better signal to decode RDS.

If you're near the transmitter, the problem could be too much gain. Try turning off hardware AGC and turning down the gain slider. If that doesn't improve the SNR, then try moving your antenna somewhere else.