gqrx-sdr/gqrx #311: cmake "make install" does not install icon and .desktop file

csete, 2015-12-09:


xand79, 2015-12-16:
Well, I have install it too with the same problem on Xubuntu 14.04 LTS. After deinstall previous version. Via 'apt-get install gqrx-sdr'. And I can't see icons at buttons on the bar (but buttons are working). Where to add it if I can?
Thanks for your great job :)

csete, 2015-12-16:
> And I can't see icons at buttons on the bar (but buttons are working)

That sounds like a different issue. I think you need to install QtSvg module. Are you sure you use version 2.4 from the new PPA?

xand79, 2015-12-17:
Help > About Gqrx: "This is Gqrx 2.4". Yes. It's from PPA. And I have libqt4-svg 4:4.8.5+git192-g085f851+dfsg-2ubuntu4.1 installed... Maybe I need other version?

csete, 2015-12-17:
Actually, yes. Gqrx in the PPA has been linked against Qt 5 so you will need the libqt5svg package. I will try fixing the package so that Qt SVG also becomes a runtime dependency (I thought it was already done)

xand79, 2015-12-17:
Big thanks for a hint! I have installed "libqt5svg5 5.2.1-1" and now can see all icons!

pinkavaj, 2022-11-07:
I cannot reproduce this error any more. @csete Can you or anyone else?

If no reponse or cannot be reproduced I'm suggesting to close this at 2022 year end.

vladisslav2011, 2022-11-07:
It is already fixed: https://github.com/gqrx-sdr/gqrx/blob/master/CMakeLists.txt#L270-L277

pinkavaj, 2022-11-07:
@argilo Seem like we can close this one, but I likely does not have the permissions. I can try to do more clean-up in the tickets, if it is not considered contraproductive, what is the suggested workflow, should I tag you in the others ticket considered to close too?

argilo, 2022-11-07:
Fixed in #1074.

argilo, 2022-11-07:
> should I tag you in the others ticket considered to close too?

Yes, if there are other tickets that should be closed, just tag me and I'll close them. Thanks for your help!

closed 2022-11-07 by argilo