gqrx-sdr/gqrx #313: Mode doen't switch to "WFM (oirt)" in boolmark.

kukabu, 2015-12-17:
Hello

When I choose bookmark with "WFM (oirt)" mode doesn't switch to.

csete, 2015-12-17:
There was a short period where the mode names were different and your bookmark was created with that mode name, which now no longer exists. So just update your bookmark or recreate it and it should work again.

kukabu, 2015-12-17:
No, names match. It creates after e2497fde038d7c8d05c77e6685e1464e86f03653.
```
$ fgrep oirt .config/gqrx/bookmarks.csv
   549750000; ТВК30 РОССИЯ 1 САМАРА    ; WFM (oirt)          ;     200000; TV
```

csete, 2015-12-24:
Actually, it looks like the correct mode is selected, but the mode shown in the mode selector is not updated. Is that also what you observe?

csete, 2015-12-24:
I believe this is fixed now, but feel free to reopen if there are still issues.

closed 2015-12-24 by csete

kukabu, 2015-12-25:
Thanks! Fixed.