gqrx-sdr/gqrx #317: Feature request: LRPT decode

xand79, 2015-12-25:
Maybe it will be great deal to add decode mode for receive LRPT data from satelites?
I found meteor_soft_symb_rtl.grc but it runs tooooo sloooowwwww on my notebook. And here is another problem: I couldn't record raw I/Q data in raw sound file. When I'm chosing decode mode "Raw I/Q" Gqrx waits for a second - and falls me to "Narrow FM". Okay, I can write full I/Q stream, but i dont't need 1 MHz wide, just 80 or 72 KHz for Meteor-M N2... And my dongle with R820T tuner didn't want to work with sample rate less then 1 024 000, I can set manually, but it slows all my Ubuntu system.
So, what I want as a user: to set filter width to 80 KHz, set mode to raw i/q, and record a raw sound file. Well, better to ges *.s file to put it in LRPToffLineDecoder by Oleg, but as minimum -  just raw narrowband i/q file.

csete, 2015-12-25:
Hi,

I'm not sure what you are asking for exactly, but a full LRPT decoder is out of the scope of gqrx, at least for now. Raw I/Q mode has not been implemented yet and I don't know when I will have the time to do it.

Note that in gqrx 2.4 we have introduced the input decimator which allows you to decrease the input sample rate while still taking advantage of high rate sampling in the hardware. For example, if you set the RTL dongle to 2.4 MHz you can decimate by 32 to get 75 kHz sample rate that you can save to I/Q file.

xand79, 2015-12-26:
Decimator! Of course! Sorry, I have crashes some times while trying to open device properties, and was using .conf file manually. But this morning I got it! Great, I'll try to decimate input.

xand79, 2015-12-29:
Hello again,
I've tried to record raw i/q with different combinations of frequency and decimation to get 72 and 80 KHz, and seems like all fine. But I don't know in what format I got *.raw records. Is it 2-channel 32-bit float or PCM?
I have found some examples in web, good raw records from Meteor-M, and its all in float format... And when I'm opening it in Audacity as raw 32-bit 2-channel sound it looks like all data is above 0 dc (between 0 and max amplitude) in both channels... maybe it's strange, but...
What's more, I've opened Gqrx raw records as PCM ('cause float produces too little amp), then got shifted all with nyquist filter (sum s 0.5), and got something like what I want, then saved it as 32-bit float. But it was wrong, and image decoder has saw nothing, even a noise...
Hm, still thinking what to do.

csete, 2015-12-29:
It's raw floating point IQ samples. Exact same as you get when you send gr-complex to a file sink in gnuradio.

xand79, 2015-12-30:
Hm, okay, I'm walking to find my mistakes.

csete, 2016-05-20:
Anything left to do here? I mean aside from LRPT decoder which is out of scope of gqrx.

troffasky, 2016-07-02:
Perhaps retitle this feature request to "mechanism for handing off to external decoder", or create a feature request like that.

There are of course a number of ways in which external decoders can be used. Some take a continuous stream of demodulated "audio" [eg dsd] and some [like I assume LRPT] would be run until you have the images you want [I could be completely wrong here, of course]. A way of handling both scenarios would be useful.

csete, 2017-03-19:
I doubt I will be able to spend any significant time on new development of gqrx as I would rather work on a new SDR which has lower complexity (in terms of dependencies) and adds transmit support.

csete, 2017-03-19:
By the way @troffasky - there is already an issue for adding a data decoder interface, see #43 
As you see, it has not got high priority.

closed 2017-03-19 by csete