gqrx-sdr/gqrx #318: build on Mac OS X is broken since commit 55cdadd

thomasgi, 2015-12-28:
When compiling GQRX on Mac OS X (10.9.5) using a macports environment following error occurs:

In file included from ../src/qtgui/dockrds.cpp:26:
./ui_dockrds.h:70:21: error: no member named 'setSizeAdjustPolicy' in 'QScrollArea'
        scrollArea->setSizeAdjustPolicy(QAbstractScrollArea::AdjustToContentsOnFirstShow);
        ~~~~~~~~~~  ^
./ui_dockrds.h:70:62: error: no member named 'AdjustToContentsOnFirstShow' in 'QAbstractScrollArea'
        scrollArea->setSizeAdjustPolicy(QAbstractScrollArea::AdjustToContentsOnFirstShow);
                                        ~~~~~~~~~~~~~~~~~~~~~^
2 errors generated.
make: *** [dockrds.o] Error 1

Macports uses QT 4.8.7-1, which does not have a method setSizeAdjustPolicy in QScrollArea or an enum QAbstractScrollArea::AdjustToContentsOnFirstShow according to the official QT 4.8 documentation.

csete, 2015-12-28:
It should be fixed now, but let me know if it didn't work.

thomasgi, 2015-12-28:
Yes, the problem is fixed now. Thank you for that super fast support.

Best regards
Thomas

closed 2015-12-28 by csete