gqrx-sdr/gqrx #319: gqrx crash with audio demodulation

vannierl, 2015-12-31:
Hello everybody. 
After compiling gqrx 2.4, i just wanted to made a test. When i started gqrx (in console mode), i have the nice spectrum and waterfall, but as soon as i wanted to use a demodulation, it crash without any warning,  and the terminal window with it. did anyone have an idea? thanks for help. loic vannier HB9EID.

my system:
raspberry pi v2 model B. 
OS: Raspbian.
messages when starting gqrx on terminal:
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using Volk machine: neon_hardfp_orc
 *** alpha: 0.242535
gr::log :INFO: audio source - Audio sink arch: alsa
BookmarksFile is /home/pi/.config/gqrx/bookmarks.csv
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner

vannierl, 2016-01-01:
I was going further in order to try to debug my problem: hereafter is the complete crash code. it seems to be caused by a variable from the sound management ALSA. I use the default sound output from my raspberry (bcm2835 ALSA (Alsa mixer)
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
gr::log :INFO: audio source - Audio sink arch: alsa
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
gr::log :ERROR: audio_alsa_sink0 - [default]: snd_pcm_sw_params: Invalid argument
terminate called after throwing an instance of 'std::runtime_error'
  what():  audio_alsa_sink
Aborted

csete, 2016-01-01:
Let's keep the discussion only one place:
https://groups.google.com/d/msg/gqrx/AUKVVzIThj4/HYwKimu4CwAJ

closed 2016-01-01 by csete

vannierl, 2016-01-01:
the problem is solved. i have updated my post on google groups with the
solution.
thank for your help. happy new year.