gqrx-sdr/gqrx #324: Crash of GQRX 2.5.1 Immediately After Starting on Mac OS X 10.11.4 Beta

fmaxwell, 2016-01-16:
## GQRX 2.5.1 Mac OS X binary crashes immediately upon startup under OS X 10.11.4 Beta.  No GUI is displayed prior to crash.

_Note:  I have never tried to run GQRX on prior versions of OS X but believe that this is unrelated to the 10.11.4 Beta version of OS X._

I Installed the GQRX binary to the Applications folder.  I start GQRX, an icon appears in dock and bounces a few times and then disappears.  Then a crash report appears (see below) which states:

`Application Specific Information:
terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
abort() called`

The crash repeats regardless of which audio output device I select in settings prior to running GQRX:

Starting from the command line with a -r option reveals the following:

	Freds-Mac-Pro:MacOS Fred$ ./gqrx -r
	Mac OS; Clang version 7.0.0 (clang-700.1.76); Boost_105900; UHD_003.009.001-0-unknown

	gr-osmosdr v0.1.4-56-g1f2511dd (0.1.5git) gnuradio 3.7.8.1
	built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
	Using Volk machine: sse4_2_64_orc
	INFO: Audio sink arch: osx
	ERROR: 

	A unique output audio device name matching the string 'built-in' was not found.

	The current known output audio device names are:
	    Built-in Output
	    Built-in Line Output
	    Built-in Digital Output
	    Audioengine D1
	    AudioQuest DragonFly

	libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
	Abort trap: 6
	Freds-Mac-Pro:MacOS Fred$ 

#### Notes:
##### 1. There are three audio device names containing the string "Built-in".
##### 2.  If the search for "built-in" is case-sensitive, it will not match any of them.

#### Crash report follows:

	Process: gqrx [5352]
	Path: /Applications/Gqrx.app/Contents/MacOS/gqrx
	Identifier: dk.gqrx.www
	Version: 2.4 (0)
	Code Type: X86-64 (Native)
	Parent Process: ??? [1]
	Responsible: gqrx [5352]
	User ID: 501

	Date/Time: 2016-01-16 02:23:00.988 -0500
	OS Version: Mac OS X 10.11.4 (15E27e)
	Report Version: 11
	Anonymous UUID: 716CAA91-EA3A-2F6A-3A16-1B4E6968B7EA

	Time Awake Since Boot: 84000 seconds

	System Integrity Protection: enabled

	Crashed Thread: 0 Dispatch queue: com.apple.main-thread

	Exception Type: EXC_CRASH (SIGABRT)
	Exception Codes: 0x0000000000000000, 0x0000000000000000
	Exception Note: EXC_CORPSE_NOTIFY

	Application Specific Information:
	terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
	abort() called

	Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
	0 libsystem_kernel.dylib 0x00007fff97662f26 __pthread_kill + 10
	1 libsystem_pthread.dylib 0x00007fff8e7524ec pthread_kill + 90
	2 libsystem_c.dylib 0x00007fff972fd6e7 abort + 129
	3 libc++abi.dylib 0x00007fff9d49cf81 abort_message + 257
	4 libc++abi.dylib 0x00007fff9d4c2a2f default_terminate_handler() + 243
	5 libobjc.A.dylib 0x00007fff8cbf84a6 _objc_terminate() + 124
	6 libc++abi.dylib 0x00007fff9d4c019e std::__terminate(void ()()) + 8
	7 libc++abi.dylib 0x00007fff9d4bfc12 __cxa_throw + 121
	8 libgnuradio-audio.3.7.8.1.dylib 0x000000010805d9de gr::audio::osx_sink::setup() + 2910
	9 libgnuradio-audio.3.7.8.1.dylib 0x000000010805bdf8 gr::audio::osx_sink::osx_sink(int, std::__1::basic_string, std::__1::allocator > const&, bool) + 552
	10 libgnuradio-audio.3.7.8.1.dylib 0x000000010805bad9 gr::audio::osx_sink_fcn(int, std::__1::basic_string, std::__1::allocator > const&, bool) + 57
	11 libgnuradio-audio.3.7.8.1.dylib 0x00000001080281b7 gr::audio::sink::make(int, std::__1::basic_string, std::__1::allocator >, bool) + 1687
	12 dk.gqrx.www 0x000000010781541e receiver::receiver(std::__1::basic_string, std::__1::allocator >, std::__1::basic_string, std::__1::allocator >, unsigned int) + 3374
	13 dk.gqrx.www 0x0000000107803d20 MainWindow::MainWindow(QString, bool, QWidget) + 1072
	14 dk.gqrx.www 0x00000001077fe8e0 main + 4352
	15 libdyld.dylib 0x00007fff997ad5ad start + 1

	Thread 1:
	0 libsystem_kernel.dylib 0x00007fff97663602 __workq_kernreturn + 10
	1 libsystem_pthread.dylib 0x00007fff8e74f578 _pthread_wqthread + 1283
	2 libsystem_pthread.dylib 0x00007fff8e74d341 start_wqthread + 13

	Thread 2:: Dispatch queue: com.apple.libdispatch-manager
	0 libsystem_kernel.dylib 0x00007fff97663f1a kevent_qos + 10
	1 libdispatch.dylib 0x00007fff9c0b4099 _dispatch_mgr_invoke + 216
	2 libdispatch.dylib 0x00007fff9c0b3d01 _dispatch_mgr_thread + 52

	Thread 3:
	0 libsystem_kernel.dylib 0x00007fff97663602 __workq_kernreturn + 10
	1 libsystem_pthread.dylib 0x00007fff8e74f578 _pthread_wqthread + 1283
	2 libsystem_pthread.dylib 0x00007fff8e74d341 start_wqthread + 13

	Thread 4:
	0 libsystem_kernel.dylib 0x00007fff97663602 __workq_kernreturn + 10
	1 libsystem_pthread.dylib 0x00007fff8e74f578 _pthread_wqthread + 1283
	2 libsystem_pthread.dylib 0x00007fff8e74d341 start_wqthread + 13

	Thread 5:
	0 libsystem_kernel.dylib 0x00007fff9765cfe2 semaphore_timedwait_trap + 10
	1 libdispatch.dylib 0x00007fff9c0b9b9b _dispatch_semaphore_wait_slow + 149
	2 libdispatch.dylib 0x00007fff9c0b3c4f _dispatch_worker_thread + 148
	3 libsystem_pthread.dylib 0x00007fff8e74f99d _pthread_body + 131
	4 libsystem_pthread.dylib 0x00007fff8e74f91a _pthread_start + 168
	5 libsystem_pthread.dylib 0x00007fff8e74d351 thread_start + 13

	Thread 0 crashed with X86 Thread State (64-bit):
	rax: 0x0000000000000000 rbx: 0x0000000000000006 rcx: 0x00007fff58405768 rdx: 0x0000000000000000
	rdi: 0x000000000000160b rsi: 0x0000000000000006 rbp: 0x00007fff58405790 rsp: 0x00007fff58405768
	r8: 0x0000000000000004 r9: 0x00007fff9d4c3a53 r10: 0x0000000008000000 r11: 0x0000000000000206
	r12: 0x00007fff584058f0 r13: 0x0000000000000006 r14: 0x00007fff792cb000 r15: 0x00007fff584057d0
	rip: 0x00007fff97662f26 rfl: 0x0000000000000206 cr2: 0x00007fff7bb6d008

	Logical CPU: 0
	Error Code: 0x02000148
	Trap Number: 133

#### Note:  Deleted "Binary Images" section for brevity.

### Hardware configuration from crash report:

	Model: MacPro5,1, BootROM MP51.007F.B03, 6 processors, 6-Core Intel Xeon, 3.46 GHz, 24 GB, SMC 1.39f11
	Graphics: ATI Radeon HD 5770, ATI Radeon HD 5770, PCIe, 1024 MB
	Memory Module: DIMM 1, 8 GB, DDR3 ECC, 1333 MHz, 0x802C, 0x33364A534631473732505A2D3147364B3120
	Memory Module: DIMM 2, 8 GB, DDR3 ECC, 1333 MHz, 0x802C, 0x33364A534631473732505A2D3147364B3120
	Memory Module: DIMM 3, 8 GB, DDR3 ECC, 1333 MHz, 0x802C, 0x31384A53463147373250445A2D3147364531
	AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8E), Broadcom BCM43xx 1.0 (5.106.98.100.24)
	Bluetooth: Version 4.4.4d11 16720, 3 services, 27 devices, 1 incoming serial ports
	Network Service: Ethernet 1, Ethernet, en0
	PCI Card: pci1b4b,9182, AHCI Controller, Slot-2
	PCI Card: ATI Radeon HD 5770, Display Controller, Slot-1
	PCI Card: PXS4, USB eXtensible Host Controller, Slot-4
	Serial ATA Device: HL-DT-ST DVD-RW GH41N
	Serial ATA Device: ATAPI iHES208 2
	Serial ATA Device: SSD2SC240GM2DH16T-T, 240.06 GB
	Serial ATA Device: Hitachi HUS724040ALE640, 4 TB
	Serial ATA Device: HGST HDS724040ALE640, 4 TB
	Serial ATA Device: Samsung SSD 840 EVO 1TB, 1 TB
	USB Device: USB Bus
	USB Device: USB Bus
	USB Device: AudioQuest DragonFly
	USB Device: USB Bus
	USB Device: BRCM2046 Hub
	USB Device: Bluetooth USB Host Controller
	USB Device: USB Bus
	USB Device: USB Bus
	USB Device: USB Bus
	USB Device: 5130
	USB Device: USB 2.0 Bus
	USB Device: Hub
	USB Device: Miscellaneous Device
	USB Device: USB2.0 Hub
	USB Device: Kensington Expert Mouse
	USB Device: Matias Keyboard
	USB Device: Apple Cinema HD Display
	USB Device: USB 2.0 Bus
	USB Device: USB 3.0 Bus
	USB Device: USB-Serial Controller D
	USB Device: Audioengine D1
	USB Device: RTL2838UHIDIR
	FireWire Device: built-in_hub, Up to 800 Mb/sec
	Thunderbolt Bus:

csete, 2016-01-16:
Ok, thanks for the report. I will see if I can find a workaround for the next build, but I am not a mac developer and have pretty much no idea what I am doing there.

fmaxwell, 2016-01-16:
I'm an embedded systems guy, so I can't offer much beyond testing.  But I'm very willing to do that.  If you have versions with debug code in them, I'll be glad to try them and provide you with all of the results.

You've got the best RTL SDR package for the Mac and I'd like to get it running on mine and help you make it work for others, too.

csete, 2016-01-16:
Thanks. I have a pretty good idea what goes wrong, so I think I will be able to trigger the error on my mac also.

csete, 2016-01-16:
This is very strange... I do not have any device named 'built-in' either but when I run 'gqrx -r' it prints for me:

INFO: Audio sink arch: osx
INFO:

Using output audio device 'Built-in Output'.
  ... which is the current default output audio device.
  Changing the default output audio device in the System Preferences will
  result in changing it here, too (with an internal reconfiguration).

It makes no sense to me why it tries to open 'built-in' for you. Do you perhaps have something set up to say 'built-in' is the default? Can you try selecting a different default in the Sound settings?

fmaxwell, 2016-01-17:
I've tried selecting every output device I have, restarting GQRX after each.  It crashes every time.

I disconnected both of my audio DACs (Audioengine D1 and AudioQuest FireFly) and tried GQRX after selecting "Built-In Digital Out" as my output.  Still crashes.

I even unloaded "AudioSwitcher," a taskbar utility for quickly changing sound input/output devices and repeated the experiments.  Again, no joy.

Anything else I can do to gather more information for you?

fmaxwell, 2016-01-17:
Follow-up:  I created a brand new user account on my Mac Pro, giving the user admin privileges.  I downloaded a fresh copy of GQRX under that account, extracted the .app file, and ran it.  The results were the same.  So the crashes are not due to some interaction with an app that is auto-starting for my original user account, or some user-specific directories/files for that account.

csete, 2016-01-17:
Ok, thanks for the info. I'll see if I can find something.

csete, 2016-01-17:
It looks like the audio_sink block throws this exception when can not find a *unique* device that has "built-in" in its name, i.e. there must be exactly one such device. 
I tried to come up with a quick-fix but failed, so I don't have any solution now, sorry.

fmaxwell, 2016-01-17:
That's a disappointment (but what I suspected at the beginning).  Looks like I'm stuck running Windows or Linux on another box (since there really aren't any other viable alternatives on OS X). 

But I really appreciate all of your efforts to find a solution to this and your generosity in making GQRX available as open source.

Kirk-Ireson, 2016-01-23:
I'm having this problem too (OSX ver. 10.10.1), and csete, I would like to echo the gratitude put forth by fmaxwell as to your efforts on this software! 

Just a little more info in trouble-shooting this issue: I have to go back to gqrx-2.2.0.dmg for the "latest" version of the available *.dmg files which works for me (Gqrx-2.4-1.dmg, Gqrx-2.5-1.dmg, and Gqrx-2.5.2-1.dmg all fail with this error).  Using "Audio MIDI Setup.app" (part of OSX) I created an "aggregate device" named "built-in" (all lower case) and it still fails (this is from Gqrx-2.5.2-1.dmg):
```
Mac OS; Clang version 7.0.0 (clang-700.1.76); Boost_105900; UHD_003.009.001-0-unknown

gr-osmosdr v0.1.4-56-g1f2511dd (0.1.5git) gnuradio 3.7.8.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy
Using Volk machine: sse4_2_64_orc
INFO: Audio sink arch: osx
ERROR:

A unique output audio device name matching the string 'built-in' was not found.

The current known output audio device names are:
    Built-in Output
    Built-in Line Output
    Built-in Line Output
    Built-in Digital Output
    Soundflower (2ch)
    Soundflower (64ch)
    built-in

libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
Abort trap: 6
```

csete, 2016-01-23:
Thanks for the info. The problem seems not to be upper/lower case but that there are many devices that have "built-in" in the name, so the audio backend raises an exception because it does not know which one to choose. I think the intended behavior was that it should always choose the default one, but for some reason it does not work.

csete, 2016-01-24:
Hi guys,

I have uploaded a new 2.5.2-2 bundle, which includes a small change that I hope fixes this issue. You can download the dmg here:
https://github.com/csete/gqrx/releases/tag/v2.5.2

Can you please try it and let me know if it works any better? Thanks.

Kirk-Ireson, 2016-01-25:
Thanks! We're getting closer. With "gqrx -r' it starts the 'Configure I/O devices' window now (it hangs with the pinwheel for about 6 seconds, then I can interact with it). If I leave 'Audio output' device set to 'Default' the program seems to launch and run fine, but I don't have audio. If I select any other device, like 'Built-in Output', the program crashes, here's an output of me selecting something other than 'Default':
```
$./gqrx -r
Mac OS; Clang version 7.0.0 (clang-700.1.76); Boost_105900; UHD_003.009.001-0-unknown

gr-osmosdr v0.1.4-56-g1f2511dd (0.1.5git) gnuradio 3.7.8.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy
Using Volk machine: sse4_2_64_orc
INFO: Audio sink arch: osx
INFO:

Using output audio device 'Built-in Output'.
  ... which is the current default output audio device.
  Changing the default output audio device in the System Preferences will
  result in changing it here, too (with an internal reconfiguration).

BookmarksFile is /Users/koick/.config/gqrx/bookmarks.csv
Number of audio input devices: 4
Number of audio output devices: 6
[waits here for selection, then outputs]
gr-osmosdr v0.1.4-56-g1f2511dd (0.1.5git) gnuradio 3.7.8.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
libc++abi.dylib: terminating with uncaught exception of type std::invalid_argument: cannot disconnect edge multiply_const_ff0:0->audio_osx_sink0:0, not found
Abort trap: 6
```

csete, 2016-01-25:
Ok, that's good news.
This was just a hack and we still need to find a proper solution, but I'm glad that at least it works now with the default device.

fmaxwell, 2016-01-25:
Sorry for the delay in getting back to you.  I had similar results to those reported by Kirk.  As long as I select the 'Default' audio device the first time, it works!  

**If I bring up the 'IO Devices'configuration menu and try to switch the device after GQRX is running, I can select another output device.**  No crash.  

That said, even if I select a particular audio device that differs from the default, GQRX will start 'following' the system default device settings.  So I use AudioSwitcher to change output devices and GQRX follows the default, which is fine for me.

BTW:  I highly recommend AudioSwitcher:
https://itunes.apple.com/us/app/audioswitcher/id561712678?mt=12

That's how I normally switch audio devices, so this version of GQRX is working great for me.

csete, 2016-01-25:
Hmm, it's so strange that it behaves so differently from the iMac I have. On that I can switch to other devices without any issues and gqrx will use them. Anyway,I'm glad that at least it works to some extent.

Kirk-Ireson, 2016-01-25:
@csete perhaps that's because your iMac only has one 'built-in' device, where as ours has more than one? (I'm actually running a Hackintosh PC)

csete, 2016-01-25:
Yes, although my imac only has one built-in device, I have plugged USB sound cards in for testing.

fmaxwell, 2016-01-26:
> Hmm, it's so strange that it behaves so differently from the iMac I have. On that I can switch 
> to other devices without any issues and gqrx will use them. Anyway,I'm glad that at least it 
> works to some extent.

I think that it may be working the same, but we are talking at cross-purposes.  

This is what happens, in order:

1.  I select "Digital Output" in my System Settings in OS X.
2.  I bring up GQRX, selecting "Default" for the audio device.
3.  It plays through that default audio device (Digital Output to AV receiver and speakers).
4.  If I bring up the GQRX *IO Devices* menu (after GQRX is running) and select "AudioQuest FireFly", it plays through that (into my headphones).
5.  I select "Internal Speakers" in my System Settings, GQRX starts playing through the internal speakers even though I had selected "AudioQuest FireFly"
6.  From that point forward, whenever I change the system default audio device, GQRX plays through the selected device.

So, it appears that the *IO Devices* menu is a one-time selection, but if you change the system settings, GQRX reverts to the default system audio device.

I hope that made things clearer rather than less so.

P.S.  None of the above is a complaint.  I'm perfectly happy with the way that GQRX working.  In order to use SDR# while GQRX was not working, I had been dual-booting into Windows 10.  To say that I was unimpressed with Windows 10 would be putting it mildly.

csete, 2016-01-26:
Ah ok, I understand now. So, it looks like that the audio backend, which is in gnuradio, switches automatically when the system default is changed.

fmaxwell, 2016-01-26:
Yes, and that causes me no problems at all.  I just set it to default anyway.  The rest of the stuff I did up there was just to provide you with more complete debug information.

fmaxwell, 2016-11-19:
It appears that this problem has reappeared in the latest release:

 ./gqrx -r
gr-osmosdr 5e71de7f (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp hackrf rfspace airspy soapy redpitaya 
INFO: Audio sink arch: osx
ERROR: 

A unique output audio device name matching the string 'built-in' was not found.

The current known output audio device names are:
    Built-in Output
    Built-in Line Output
    Built-in Digital Output
    AudioQuest DragonFly

libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
Abort trap: 6
Freds-Mac-Pro:MacOS Fred$ 
Freds-Mac-Pro:MacOS Fred$ ./gqrx -r
gr-osmosdr 5e71de7f (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp hackrf rfspace airspy soapy redpitaya 
INFO: Audio sink arch: osx
ERROR: 

A unique output audio device name matching the string 'built-in' was not found.

The current known output audio device names are:
    Built-in Output
    Built-in Line Output
    Built-in Digital Output
    AudioQuest DragonFly

libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
Abort trap: 6

csete, 2016-11-19:
I may have forgotten to apply or commit a patch that I made for this issue, sorry.

webmariner, 2016-12-12:
Hi there - I'm running v2.6 and I noticed that it kept crashing until I took my headphones out after reading this thread. Then started the app without them, plugged them back in and all works fine. Thanks for a great app! :-)

acceleroto, 2017-01-07:
I'm seeing the crash too. Pulling my headphones didn't change anything. These are the audio output devices gqrx -r lists on my 2010 Mac Pro running OS X 10.11.6.
    AirPlay
    Built-in Output
    Built-in Line Output
    Built-in Digital Output
    Soundbooth 2.0

It's the same unique built-in problem as before.

Thanks!

vitalyga, 2017-02-01:
Hi there, is there any advance on this issue? I have the same problem with "built-in output". No matter what I choose, gqrx crashes with error message "cannot disconnect edge multiply_const_ff0:0->pa_sink0:0, not found". I have pulseaudio installed and running. Any suggestions?

dsward2, 2017-04-05:
I have a possible work-around on the 'built-in' unique output audio device name error for Mac Pro.  

Launch Terminal.app and enter these two commands. The first command sets the device name to one of the "current known output audio device names".  The second command launches the app.

export GR_CONF_AUDIO_OSX_DEFAULT_OUTPUT_DEVICE="Built-in Line Output"

/Applications/Gqrx.app/Contents/MacOS/gqrx 

I believe the export could be made permanent in ~/.profile

I don't have an SDR device yet, so I haven't actually tested it.  Amazon will deliver an SDR later today.

dsward2, 2017-04-06:
I've received a RTL-SDR dongle now.  With the environment variable export mentioned in the previous message to prevent the app crash, gqrx is receiving signals from the dongle - but so far, it is not producing any audio output through the Mac Pro audio output ports.  I've tried all output ports listed in the Configure I/O Devices panel.

I may try building the source and setting some breakpoints.  I'll update this thread if any more info about Mac Pro audio is found.

LongnoseRob, 2017-04-06:
Just some simple things to check:
Which demodulator have you set?
Is the squelch open?
Have you tried adjusting the audio gain at the bottom of the audio spectrum window?

(Sent from my mobile device)

dsward2, 2017-04-06:
LongnoseRob - gqrx runs okay on a MacBook, but the same settings (except for the audio settings) on Mac Pro haven't produced any audio output yet.  I've mostly tested broadcast FM (wide FM) so far.

I'm trying several different SDR applications on macOS.  So far, gqrx is the easiest to get up and running on a Mac - except for the Mac Pro, but that looks fixable.

dsward2, 2017-04-06:
The Mac Pro problem is reproducible in the gnuradio companion app installed with MacPorts, with the same message about the current known output audio device names.

I'm checking gnuradio's "Audio Sink" component for more info.  It has a property for a device name.

csete, 2017-04-06:
@dsward2 

Thanks for the update. If you have GNU Radio SDK installed you are pretty close to being able to build gqrx. Since version 2.6 there is an option to build using Portaudio as backend so you could try that. I'm sure Portaudio is also available through macports.

Let me know if you need more info.

dsward2, 2017-04-07:
csete - Thanks for a nice app.

Gqrx audio is working on my Mac Pro now.  Although launching the main bundle icon on the Finder desktop still fails on Mac Pro, I found a convenient workaround.  I used the Automator application in /Applications/Utilities to create a new Application work flow, added a Run Shell Script command with a path like "/Applications/gqrx.app/Contents/MacOS/gqrx", and save it as a new application named "Start Gqrx.app".  Then double-click on that app to launch gqrx.

The other important detail for Mac Pro audio is to set the Receiver Options Squelch value to -10.4 dBFS or lower.

Gymnae, 2017-05-03:
I'm encountering the same error: 

```
A unique output audio device name matching the string 'built-in' was not found.

The current known output audio device names are:
    Built-in Output
    Built-in Line Output
    Built-in Line Output
    Built-in Digital Output
    Sound Blaster Tactic(3D) Sigma
    Soundflower (2ch)
    Soundflower (64ch)

libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: audio_osx_sink::setup
Abort trap: 6
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...22 completed.
```

The 2.5.2-3 package starts, the release 2.6-1 from october 2016 won't

csete, 2017-05-03:
Hi,

I am considering rewriting the audio backend (again) and use the QAudio API instead. Before I do that I would like you to test it to see if it works.

You can test it using an existing application that already uses QAudio:

1. Go to http://rfspace.com/RFSPACE/CloudIQ.html
2. Download "RemoteSDRclient Application for Mac OS v1.20"
3. Select audio device (Setup -> Sound card)
4. Connect to a public SDR server (Setup -> Network)
5. Start the SDR, if the connection to server is OK you should be hearing audio

fmaxwell, 2017-05-03:
Hi,

It showed the following audio output devices in my system:

ODAC
Built-In Output
Built-In Line Output
Built-in Digital Output
Audioquest Dragonfly

I verified that each the first, second, and fifth worked.  I have nothing connected to the other two, but selecting them resulted in no audio.

Please let me know if I can be of further assistance.

csete, 2017-05-03:
Hello @fmaxwell 

Thank you for the quick test. This gives me confidence that I should switch to this backend. It will also simplify the code and reduce the number of external dependencies, so it's a win-win :)

It needs some work in gqrx, so probably a week or two, but I can post an update when the implementation done. I can also make a bundle for testing if necessary.

dranch, 2017-05-04:
Hello Alex,
What version of Qt would you be requiring to have installed to support the new audio backend?  I hope it would be <= Qt 5.6 (from a third party repo) or that would knock out us Centos 6 / RHEL users.

csete, 2017-05-04:
Hi David,

It should work with Qt 5.5, theoretically even with 5.2; however, this Qt package has been improved a lot so at least 5.6 should be used.

Note that you can always install newer and even multiple versions of Qt using their own installer. You can even install under your home directory not affecting system wide packages.

csete, 2017-05-04:
I forgot to mention that I will introduce it as a build time option so there should be time to test it.

stromnet, 2017-06-22:
Tried with the GR_CONF_AUDIO_OSX_DEFAULT_OUTPUT_DEVICE trick above:
```
export GR_CONF_AUDIO_OSX_DEFAULT_OUTPUT_DEVICE="Built-in Line Output"
./qrx
```

but it failed, since my box has two "Built-in Line Output".
I was able to work around it using MIDI Setup and create a new Aggregate Audio Device with a custom name, and then use that in the env var. Works! :) 
(https://apple.stackexchange.com/questions/272/is-there-a-way-to-rename-a-sound-card-in-mac-os-x)

hemna, 2017-08-24:
If I export as described as above the app launches, but then freezes upon selecting the RTL SDR device.   I get no UI, it just freezes.   

I can run the rtl_tcp localhost:1234 manually from the command line and get:

>  waboring@Walters-Mac-Pro  /Applications/Gqrx.app/Contents/MacOS  ./rtl_tcp localhost:1234                                                            ✔  15:33:18
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

But launching gqrx results in no UI and force quit says gqrx is not responding.

csete, 2018-03-18:
There is a new version 2.11.1 which uses portaudio as backend and works well on Mac Pro that doesn't have the built-in audio device.

closed 2018-03-18 by csete