gqrx-sdr/gqrx #325: Feature Request: Setting offset with gqrx remote control

jbhunter21, 2016-01-18:
Hi,
I've seen this feature in SDR# where you could bookmark a center-frequency and an offset.
That would be nice to have in gqrx, especially for scanners / scripts / programs using gqrx remote control.
Instead of having the delayed caused by the receiver retuning to every frequency you are interested in, you could set a center-frequency in the bandwidth you're scanning (most useful with receivers like HackRF or other high bandwidth recivers) and then just filtering through channels / frequencies using the offset saved in the bookmarks.

An example would be when using a HackRF receiver and setting the center-frequency to 127.500MHz and then only using the offset to listen to specific channels / frequencies within that 19MHz airband (118MHz-137MHz)
That would cut away allot of delay in the scanning because the receiver does not tune to different frequencies, it stays put on the center 127.500MHz at all time.

csete, 2016-01-18:
Gqrx should already do this automatically if the new frequency fits within the bandwidth - or perhaps I am misunderstanding what you are asking for.

jbhunter21, 2016-01-18:
Yes, now that I think about it, it actually does that. Sorry for this unnecessary feature request. :P

csete, 2016-01-18:
Hey, no problem. I am closing the request but feel free to let use know if it is not working as it should or as you expect. Also, there is a limit at the edges where it starts moving the center frequency and we can move that limit if it is too narrow.

closed 2016-01-18 by csete