gqrx-sdr/gqrx #326: install gqrx from pybombs gnuradio 3.7.10 issue

andrew771211, 2016-01-18:
Hi all
 I installed gnuradio 3.7.10 by pybombs but gqrx fails:
and@and-HP-250-G4-Notebook-PC:~/pybombs$ sudo ./app_store.py
[sudo] password for and: 
Settled on prefix: /home/and/target
Initializing environmental variables...
None
Loading recipes ...
(python2:2810): IBUS-WARNING **: The owner of /home/and/.config/ibus/bus is not root!
installing gqrx
/home/and/target
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of libpulse-dev (4.0) is >= than 1.1
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of libqt4-dev (4.8.5) is >= than 4.6.2
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of cmake (2.8.12) is >= than 2.8.3
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of cmake-data (2.8.12) is >= than 2.8.3
Installing packages:
* gqrx
Installing from source: gqrx
PyBombs.sysutils - INFO - monitor: Tried command 0 time(s)
Cloning into 'gqrx'...
remote: Counting objects: 6504, done.
remote: Total 6504 (delta 0), reused 0 (delta 0), pack-reused 6504
Receiving objects: 100% (6504/6504), 4.62 MiB | 224.00 KiB/s, done.
Resolving deltas: 100% (3779/3779), done.
Checking connectivity... done.
PyBombs.sysutils - INFO - shellexec_shell: Caught the quit Event
 
    CC=gcc CXX=g++ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/home/and/target $config_opt

Configuring: (100%) [=========================================================] 
Configuration failed. Re-trying with higher verbosity.
make: *** No targets specified and no makefile found.  Stop.
Build failed. See output above for error messages.

this is the gqrx.lwr file I have:
category: application
depends: gnuradio gr-osmosdr pulseaudio qt4
satisfy_deb: gqrx-sdr
source: git://https://github.com/csete/gqrx.git
gitbranch: master
inherit: empty

configure {
    qmake 
}

make {
    make -j$makewidth
}

install {
    cp src/gqrx $prefix/bin/
}

uninstall {
    rm $prefix/bin/gqrx
}
Any help is highly appreciate
thanks

csete, 2016-01-18:
The recipe seems to e using qmake, while the actual build runs with cmake. Other than that , I have no idea what is wrong. You should probably try the gnuradio list instead.

andrew771211, 2016-01-18:
Sorry, I pasted a different file from the original taken from a "old" gnuradio installation (I also tested with this file, it install gqrx but doesn't launch: the application is not installed)
this is the gqrx.lwr original:

category: application
depends: gnuradio gr-osmosdr pulseaudio qt4
satisfy_deb: gqrx-sdr
source: git://https://github.com/csete/gqrx.git
gitbranch: master
inherit: cmake

install {
    cp src/gqrx $prefix/bin/
}

uninstall {
    rm $prefix/bin/gqrx
}

csete, 2016-01-18:
Ok, but I still can't help you with pybombs.

JonZeolla, 2016-01-23:
I've run into a similar issue with a fresh install of Ubuntu 14.04.  At first I thought it was a pybombs issue but when I came here and attempted to follow the manual steps you provide (https://github.com/csete/gqrx/blob/master/README.md) I get a similar error.  Below are screenshots - first is me trying via pybombs with -v, second is me trying with your commands via the README.  

[image]
[image]

csete, 2016-01-23:
I don't see how this error is similar to the previous one.

As the error very clearly says, cmake can not find the Qt5. So, you have to tell it where to find it, if it is at all installed.

closed 2016-01-23 by csete