gqrx-sdr/gqrx #327: Main window height

rag2, 2016-01-19:
# Main window height is too much for **768** height screen
A 768 height screen is found on many laptops or cheap screens.
As of GQRX 2.5, the minimum height of the standard screen layout seems to be **787** pixels.

## Workaround
One of the control panels may be minimised, thus losing the associated display and control, and the main screen resized. However, when the panel is restored, such that 2 panels are shown, the main screen returns to 787 pixel height.

## Request
That the QGRX GUI be set such that all the functionality fits inside a 768 pixel height screen.

csete, 2016-01-19:
Thanks Robin. I think this happened with increasing the button sizes due to some constraints on OS X. I will try to fix it for the next release.

csete, 2016-01-19:
I managed to squeeze the layouts and add some scroll windows so that the window can now almost fit into 800 x 600 pixels, though still depending on the desktop theme.

[image]

closed 2016-01-19 by csete

rag2, 2016-01-19:
Alex, many thanks for that!

csete, 2016-01-19:
I had to re-increase the height a bit, otherwise it would be broken on OS X, but still below 700 pixels. I'll make a new release soon.