gqrx-sdr/gqrx #329: Feature request: implement answer for get 'dump_state' command in remote_control.cpp

ly2ss, 2016-01-23:
I want to talk to gqrx via rigctl -m 2 commands (for example xdx client), but when connected I get time-out at once because, the answer to dump_state command is not implemented. (rigctl issues it to the remote server at the very beginning),

I guess something similar is done here:
https://github.com/hexameron/rtl-sdrangelove/blob/master/plugins/channel/tcpsrc/rigctl.cpp

csete, 2016-01-23:
Ok, don't know when I will have time though.

ly2ss, 2016-01-24:
This piece of code works OK (borrowed from the link).
This is raw hack because I am not that good at programming

    else if (buffer[0] == '\\' || buffer[0] == '1')
    {
        // See dump_state in rigctl_parse.c for what this means.
        //("%"FREQFMT" %"FREQFMT" 0x%x %d %d 0x%x 0x%x\n",start,end,modes,low_power,high_power,vfo,ant)
        rc_socket->write("0\n");
        rc_socket->write("2\n");
        rc_socket->write("1\n");
        rc_socket->write("150000.000000 30000000.000000  0x900af -1 -1 0x10000003 0x3\n");
        rc_socket->write("0 0 0 0 0 0 0\n");
        rc_socket->write("150000.000000 30000000.000000  0x900af -1 -1 0x10000003 0x3\n");
        rc_socket->write("0 0 0 0 0 0 0\n");
        rc_socket->write("0 0\n");
        rc_socket->write("0 0\n");
        rc_socket->write("0\n");
        rc_socket->write("0\n");
        rc_socket->write("0\n");
        rc_socket->write("0\n");
        rc_socket->write("\n");
        rc_socket->write("\n");
        rc_socket->write("0x0\n");
        rc_socket->write("0x0\n");
        rc_socket->write("0x0\n");
        rc_socket->write("0x0\n");
        rc_socket->write("0x0\n");
        rc_socket->write("0\n");
    }

ly2ss, 2016-01-24:
Interesting: SSB is OK. But with CW, spot data puts gqrx right on the carrier frequency (zero beat).
So if the mode is CW there's should be some math (+/-) with CW/filter bandwith/2 involved ?

csete, 2016-01-24:
Not sure how this is related to the dump_state, but CW mode in gqrx is just plain LSB/USB with narrower filters.

csete, 2016-01-24:
Just to be clear, I understand what you mean with the CW issue. I have been aware of it since the beginning, but nobody has been complaining about it so I didn't change it. I will see if I can fix it easily, otherwise it will have to wait because I am very busy at the moment.

ly2ss, 2016-01-24:
Sorry for OT. It's ok Alexandru, sorry for disturbing, 73.

csete, 2016-01-24:
> Sorry for OT. It's ok Alexandru, sorry for disturbing, 73.

Hey, that's not what I meant, I can just switch off emails if I don't want to be disturbed ;-)
Just for info that I may not be able to fix it quickly.

csete, 2016-01-24:
I just pushed a bunch of changes that change the way CW is received. In CW mode the filter is now symmetric around the carrier so it should work with correct frequency.

There is still CWL and CWU. Sidetone can be changed under demod options.

I would appreciate if you could test it thoroughly to see if it works properly and that it doesn't break any other modes. Thanks.

ly2ss, 2016-01-25:
Good evening!
Confirming - works correct (other modes are OK too).
I will report if anything is wrong.
I like the way you implemented CW bandwidth tuning.

csete, 2016-01-28:
Ok, I have added the response you posted. I assume the network command is "dump_state". Can you test it and see if it works or if there is more we have to do.

ly2ss, 2016-01-28:
It does not work:
rigctl -m 2 -r localhost:4532 -vvvvv
rigctl, Hamlib 3.1~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig:rig_init called 
dummy: _init called
rig_register (1)
rig_register (2)
rig:rig_open called 
netrigctl_open called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.    
read_string(): RX 7 characters
0000    52 50 52 54 20 31 0a                                RPRT 1.         
read_string(): Timed out 2.2037 seconds after 0 chars
rig_open: error = Communication timed out 

I think it's because in line 286:  the check is made only for one byte: which is \ or 1
(that part I pasted was working couple days ago, but when I tried to replace your check with that code now I get at recompile:

/src/pybombs/src/gqrx/src/applications/gqrx/remote_control.cpp: In member function ‘void RemoteControl::startRead()’:
/src/pybombs/src/gqrx/src/applications/gqrx/remote_control.cpp:286:28: error: invalid conversion from ‘char’ to ‘const char*’ [-fpermissive]
     else if (cmdlist[0] == '\\' || cmdlist[0] == '1')

Hmm  earlier it was checking for buffer variable
Now you do: QStringList cmdlist = QString(buffer).trimmed().split(" ", QString::SkipEmptyParts);

I don't know how to repair the code now....

But yes, checking only for one byte '\' instead of whole '\dump_state' command is too simplistic, because there might be other commands starting with \

csete, 2016-01-29:
https://github.com/csete/gqrx/commit/5e28ad0b9361b4ae2d1d9aa3a47c4cf6a36e8564#commitcomment-15753050

It looks like we need two backslashes in front of dump_state, i.e. \\\dump_state.

csete, 2016-01-29:
Now it is easier to see what commands are checked, see here:
https://github.com/csete/gqrx/blob/master/src/applications/gqrx/remote_control.cpp#L286

ly2ss, 2016-01-29:
This is how it works here:
else if (cmdlist[0] == "\\dump_state")
 at least now it connects happily and responds to f commands.

But all this somehow broke  "F" command. 

Rig command: F10000000
Frequency: netrigctl_set_freq called
write_block(): TX 18 bytes
0000    46 20 31 30 30 30 30 30 30 30 2e 30 30 30 30 30     F 10000000.00000
0010    30 0a                                               0.              
read_string(): RX 7 characters
0000    52 50 52 54 20 31 0a                                RPRT 1.         
set_freq: error = Protocol error

(I am testing this with rigctl -m 2 -vvvvv)

ly2ss, 2016-01-29:
Jeeez... I am sure I pasted two slashes (it must be two, one is not enough) and this issue web site omitted one slash.

csete, 2016-01-29:
Yes, it is not easy when backslash is used for escaping characters :-)

I have just committed the change and also changed the frequency reader because it was apparently not compatible with rigctl. I hope it works now.

ly2ss, 2016-01-29:
Oh man! IT works :-) At least with xdx. Thanks!!! Tried grig, flidigi - no luck but at least I know what to do if something is bad... As you mentioned - the  support is very limited. Hence I can't complain. 73!

oddstr13, 2016-01-29:
I've been working on and off on a Python script proxy that would achieve this. Hasn't gotten it properly working yet, but it might provide some insight as to what the different parts of dump_state means.
https://gist.github.com/oddstr13/872da895e618ba5c8d60

I have been using this as a reference; http://sourceforge.net/p/hamlib/code/ci/master/tree/tests/rigctl_parse.c#l2620

ly2ss, 2016-01-29:
That's interesting thank you - proxy might be useful to change modes depending on freq... 
I was using funcube gqrx synced with ft857 via hamlib + rigctl + tlf. I remember I was missing that capability -  depending on spot frequency, change the mode.
Now having only hackrf.

csete, 2016-01-29:
Feel free to ask for additional commands that might be needed to make existing applications work. I'm sure I or somebody else will take a look at it and assess whether it make sense to add them. For example, making fldigi work sounds reasonable; however, implementing the full set of commands to make grig happy seems a bit pointless.

I have now added some debug code that will print unknown commands to the terminal. Take those and put it in a new issue here on github and we will take a look. I'm closing this issue now since the original request is now accomplished.

closed 2016-01-29 by csete