gqrx-sdr/gqrx #330: Frequency edges

ntoulasd, 2016-01-25:
While drag and scroll frequency from waterfall. Black edges appears.
Could be possible to auto change main frequency so always be in range.

I don’t know how to explain better way. Sorry.
[image]

csete, 2016-01-25:
Well, yes, that's how it works. There is no "drag to tune" functionality if that is what you are trying to do. The dragging is only intended to pan the spectrum when you are zoomed in. If you want to tune you have to click on the digits above the spectrum.

csete, 2016-01-25:
By the way, there are buttons to re-center the spectrum in the FFT settings tab.

ntoulasd, 2016-01-25:
Yes of curse, I use the buttons. 
But sometimes "drag to tune" helps to view further frequencies, while be in zoom.

csete, 2016-01-25:
But there is nothing to see beyond the border. Gqrx can only show what corresponds to the sample rate centered around the center frequency. If you want to se higher or lower frequencies you have to move the center frequency.

Sorry, but this is how it works and has always worked. I know some other SDR application allow you to drag and will retune automatically but that is no implemented in gqrx.

ntoulasd, 2016-01-25:
Ok I understand. Consider it as a feature request.
Thanks

csete, 2016-01-25:
Okay, fair enough :-)
I do like the idea.

ntoulasd, 2016-01-25:
I am glad that you like it :)

icthieves, 2017-06-27:
I also would really like the ability to "drag" the center frequency.
Not necessarily in a live-tuning way, even just being able to drag and drop the center frequency to an interesting spot and then use the digits above the spectrum for fine-tuning.
Or even right-clicking somewhere on the spectrum and having it automatically recenter there.

ankostis, 2020-11-30:
Also PgUp/PgDn keys would be nice to retune the central frequency up or down.

ankostis, 2020-11-30:
Also #/793 is relevant

lmutt, 2026-09-13:
I also think it would be very useful to tune by dragging across the spectrum. I wonder if this feature request will ever be considered...