gqrx-sdr/gqrx #331: gqrx not opening on osx

Brett99, 2016-01-26:
When I try to open the application, it says "gqrx quit unexpectedly" every time.

Brett99, 2016-01-26:
[Here is the crash log](https://github.com/csete/gqrx/files/104271/crash.txt)

Brett99, 2016-01-26:
My fault, I was missing the line '127.0.0.1 localhost' in my host file.

closed 2016-01-26 by Brett99

nate-nate, 2016-04-10:
I'm having the same issue. 
Could you explain how you added the localhost line to your host file please?

Brett99, 2016-04-10:
Edit the file at /etc/hosts

Add the following line to this file: '127.0.0.1 localhost'

nate-nate, 2016-04-10:
My terminal isn't prompting me for my admin password (i'm the only user/admin)
instead it says I'm not permitted to access my hosts file

Thanks for the help

nate-nate, 2016-04-10:
sudo nano worked - looks like I have that line in the file already!