gqrx-sdr/gqrx #332: /dev/random issue for windows build

guruofquality, 2016-01-26:
I was testing the GQRX for the new PothosSDR build. Not surprisingly it bombs on the /dev/random FIXME part. Otherwise, I don't think its that far away from working since all of the dependencies are working.

```
    if (input_device.empty())
    {
        // FIXME: other OS
        src = osmosdr::source::make("file=/dev/random,freq=428e6,rate=96000,repeat=true,throttle=true");
    }
```

I'm not sure how to work around this since there is not a random device. But perhaps a cheap substitute would be to load a file on the fly with random floats and use that instead? I'm willing to try anything out.

csete, 2016-01-26:
Technically, we could read any file even if it contains NaN values since this source is only used until the configuration is read and not while the flow graph is running - at least not as long as the user config is OK. And if the user config is bad the application will crash anyway at some point.

Alternatively, we could try using an rtl_tcp source instead. I believe that is also available without any further hardware requirements.

guruofquality, 2016-01-27:
It works!

[image]

My little patch
====
Not the best way to handle it, but basically we sub in a temp null and temp random file on repeat. I used Qt API to make it cross platform. This could be generalized better or copied into their own file. Also the wav source doesnt have a repeat option so the behaviour wouldnt be identical in that case:

```
diff --git a/src/applications/gqrx/receiver.cpp b/src/applications/gqrx/receiver.cpp
index 55254e3..d84eb48 100644
--- a/src/applications/gqrx/receiver.cpp
+++ b/src/applications/gqrx/receiver.cpp
@@ -47,6 +47,55 @@
 #include <gnuradio/audio/sink.h>
 #endif
 
+#include <QFileInfo>
+#include <QTemporaryFile>
+#include <QDataStream>
+
+static std::string get_random_file(void)
+{
+    static std::string path;
+    if (path.empty())
+    {
+        path = "/dev/random";
+        QFileInfo checkFile(QString::fromStdString(path));
+        if (!checkFile.exists())
+        {
+            //static temp file persists until process end
+            static QTemporaryFile temp_file;
+            temp_file.open();
+            path = temp_file.fileName().toStdString();
+            {
+                QDataStream stream(&temp_file);
+                for (size_t i = 0; i < 1024*8; i++) stream << qint8(rand());
+            }
+            std::cout << "Created random file " << path << std::endl;
+        }
+    }
+    return path;
+}
+
+static std::string get_null_file(void)
+{
+    static std::string path;
+    if (path.empty())
+    {
+        path = "/dev/null";
+        QFileInfo checkFile(QString::fromStdString(path));
+        if (!checkFile.exists())
+        {
+            //static temp file persists until process end
+            static QTemporaryFile temp_file;
+            temp_file.open();
+            path = temp_file.fileName().toStdString();
+            {
+                QDataStream stream(&temp_file);
+                for (size_t i = 0; i < 1024*8; i++) stream << qint8(0);
+            }
+            std::cout << "Created null file " << path << std::endl;
+        }
+    }
+    return path;
+}
 
 /**
  * @brief Public contructor.
@@ -77,8 +126,7 @@ receiver::receiver(const std::string input_device,
 
     if (input_device.empty())
     {
-        // FIXME: other OS
-        src = osmosdr::source::make("file=/dev/random,freq=428e6,rate=96000,repeat=true,throttle=true");
+        src = osmosdr::source::make("file="+get_random_file()+",freq=428e6,rate=96000,repeat=true,throttle=true");
     }
     else
     {
@@ -110,7 +158,7 @@ receiver::receiver(const std::string input_device,
 
 
     // create I/Q sink and close it
-    iq_sink = gr::blocks::file_sink::make(sizeof(gr_complex), "/dev/null", false);
+    iq_sink = gr::blocks::file_sink::make(sizeof(gr_complex), get_null_file().c_str(), true);
     iq_sink->set_unbuffered(true);
     iq_sink->close();
 
@@ -127,7 +175,7 @@ receiver::receiver(const std::string input_device,
     audio_gain0 = gr::blocks::multiply_const_ff::make(0.1);
     audio_gain1 = gr::blocks::multiply_const_ff::make(0.1);
 
-    wav_sink = gr::blocks::wavfile_sink::make("/dev/null", 2,
+    wav_sink = gr::blocks::wavfile_sink::make(get_null_file().c_str(), 2,
                                               (unsigned int) d_audio_rate,
                                               16);
 
```

Icons are missing
====
The icons are missing, which I guess you would expect. For PothosGui I basically copied all the relevant icons into the source tree that I wanted from an icon pack and installed them. The gui looks for the default theme and uses the copied icons as a fallback with ```QIcon::fromTheme``` If that helps...

csete, 2016-01-27:
Good news, thanks for the patch. I'll see if I can apply it.

Regarding the icons, they are all included in the binary, so there is nothing to install. But it needs the QtSvg module in the runtime library path.

guruofquality, 2016-01-27:
> Regarding the icons, they are all included in the binary, so there is nothing to install. But it needs the QtSvg module in the runtime library path.

Ah. I was making some assumptions. Anyway, the QtSvg is copied over (FWIW, I think we more or less use the same Qt5 DLLs). I will have to take a closer look I guess

guruofquality, 2016-01-27:
@csete You were on the right track. I needed to package iconengines\qsvgicon.dll as well. Icons look good.

guruofquality, 2016-01-27:
Let me know if should clean up the patch and create a pull request. I would probably put the ```std::string get_random_file(void)``` and ```std::string get_null_file(void)``` into its own c++ file and the changes to receiver.cc would be quite minimal. Between this issue and https://github.com/csete/gqrx/pull/333 thats it!

csete, 2016-01-27:
You are are more than welcome to send a pull request, that would make it easier for me. You can then also decide whether you want to put the new functions in their own file (sounds like a good idea though).

guruofquality, 2016-01-27:
Awesome, with this merged https://github.com/csete/gqrx/pull/335 .. I guess thats everything. I will have new installers built and uploaded today or tomorrow. Thanks!

closed 2016-01-27 by guruofquality

csete, 2016-01-27:
Nice. Where do you keep the downloads?
I will probably make a new release later this week or next week.

guruofquality, 2016-01-27:
> Nice. Where do you keep the downloads?

This is the homepage for the installer: https://github.com/pothosware/PothosSDR/wiki but downloads are actually hosted on the myriadrf downloads server: http://downloads.myriadrf.org/builds/PothosSDR/ (thanks again Andrew)

I was thinking of either creating a brief GQRX wiki page or linking to one.Just a few screenshots and some minor instructions... Although the instructions are pretty strait forward (install + click shortcut), there are some helpful tips in the tutorial as well. I think GQRX users will want follow the "download and install" and "recognizing your device" part: https://github.com/pothosware/PothosSDR/wiki/Tutorial#recognizing-your-device

> I will probably make a new release later this week or next week.

Cool. In general I try to track the most recent release tags. Unless there were just some important merges or the project doesnt really have a release tag. And then I rebuild about once a month -- whatever the state of the SDR ecosystem happens to be in.