gqrx-sdr/gqrx #336: GQRX and FCD Pro+

loewal, 2016-01-28:
Hi,
Is it a known issue that my Funcube pro+ doesn't appear under my devicelist anymore, while an rtl-adapter does?
Other programs like SDRDX work fine.
I am running the latest beta of OSX 10.11.4.

Thank you for a solution
Grtz. Loe

csete, 2016-01-28:
Hi Loe,

For me it does work and shows up as "FUNcube Dongle V2.0" in the list of devices. When did it stop working, when you upgraded gqrx or when you upgraded OS X? Are you using macports or the bundles?

In case of macports you may have to install an fcdproplus variant.

Also, which Mac do you have? There are some issues with running gqrx on Mac Pro while it seems to wirk fine on my iMac and Macbook pro.

I have done some changes in the audio area lately trying to address the issues on Mac Pro, see #324, so you could try one of the earlier releases to see when it broke.

Alex

loewal, 2016-01-28:
Hi Alex,
I am running version 2.5.2.2 (dmg).
On my late 2013 iMac and the new Macbook. They stopped working when I upgraded from OSX 10.3 to 10.4.
A few months ago compiling Gqrx within macports gave an error on qt4 and qt5, so I was happy to find your precompiled app on Sourceforge.
Even the older versions (2.4.1) give the same problem: no FCD Pro+!
I somewhere read that another user has the same problem.
I tried SDRDX: just fine! So, why did it disappear....?
Also the 2.3.1 version on this site stopped working: http://subdimensions.com/download/
Its a pity. I installed SDR# on a bootcamp partition to solve this (hopefully not permanent) problem.

Please help...
Grtz,
Loe

csete, 2016-01-28:
I have no idea why it disappeared, I'm not even sure what you mean by disappear. Do you mean that it is no longer on the list of available devices?

I can only come with some generic suggestions:

Make sure other SDR applications are quit, so that they don't block access to the FCD.

Start gqrx from the terminal and lets see what info it prints. You can do that with Gqrx.app/Contents/MacOS/gqrx

You can also try to launch on the command line using the -r option to reset configuration.

loewal, 2016-01-28:
Hi Alex,

FCD Pro+ is not on the list of available devices when I configure Gqrx. Only Airspy etc. and when I plug in a RTL-dongle it is listed,but NO FCD!
I know that option of starting Gqrx from the terminal. No effect, the same problem.
There is no other device active. And as a told you, SDRDX works fine....
Could it be a OSX beta problem?
Loe

csete, 2016-01-28:
I suppose it could be a beta problem but it is strange that SDRDX works and gqrx does not.

Regarding the terminal, I was looking for any possible error or warning messages printed during start or when the I/O configurator is loaded to see if there are any clues.

Anyway, I am working on a new funcube dongle driver that will make it easier to test and debug such issues. It is a little difficult in gqrx at the moment because there are so many layers.

loewal, 2016-01-28:
Thank you for your quick response Alex!
For the moment I am running SDR#, Studio-1-SDR and SDRconsole v2.
But...in Windows...a pity.

loewal, 2016-01-28:
I will send you a printout of the errormessage of Gqrx.
When i have switched to OSX again.

loewal, 2016-01-28:
Hi Alex,
My printout:
[gqrx.txt](https://github.com/csete/gqrx/files/108829/gqrx.txt)

and a screenshot:
[image]

loewal, 2016-01-28:
And this:
drwxr-xr-x@ 19 loewal  admin      646 29 aug  2014 .
drwxr-xr-x@  8 loewal  admin      272 28 jan 21:37 ..
-rwxr-xr-x@  1 loewal  admin  1646612 29 aug  2014 Gqrx
-rwxr-xr-x@  1 loewal  admin    13868 29 aug  2014 hackrf_cpldjtag
-rwxr-xr-x@  1 loewal  admin    13560 29 aug  2014 hackrf_info
-rwxr-xr-x@  1 loewal  admin    13564 29 aug  2014 hackrf_max2837
-rwxr-xr-x@  1 loewal  admin    13596 29 aug  2014 hackrf_rffc5071
-rwxr-xr-x@  1 loewal  admin    13776 29 aug  2014 hackrf_si5351c
-rwxr-xr-x@  1 loewal  admin    13968 29 aug  2014 hackrf_spiflash
-rwxr-xr-x@  1 loewal  admin    25612 29 aug  2014 hackrf_transfer
-rwxr-xr-x@  1 loewal  admin    25684 29 aug  2014 rtl_adsb
-rwxr-xr-x@  1 loewal  admin    19940 29 aug  2014 rtl_sdr
-rwxr-xr-x@  1 loewal  admin    26104 29 aug  2014 rtl_tcp
-rwxr-xr-x@  1 loewal  admin    24740 29 aug  2014 rtl_test
-rwxr-xr-x@  1 loewal  admin   320832 29 aug  2014 uhd_cal_rx_iq_balance
-rwxr-xr-x@  1 loewal  admin   331900 29 aug  2014 uhd_cal_tx_dc_offset
-rwxr-xr-x@  1 loewal  admin   331900 29 aug  2014 uhd_cal_tx_iq_balance
-rwxr-xr-x@  1 loewal  admin   145772 29 aug  2014 uhd_find_devices
-rwxr-xr-x@  1 loewal  admin   214464 29 aug  2014 uhd_usrp_probe
iMac-van-Loe-Walter:MacOS loewal$ ./Gqrx
Mac OS; Clang version 5.1 (clang-503.0.40); Boost_105600; UHD_003.007.002-0-unknown

gr-osmosdr v0.1.1-13-g58d95b51 (0.1.2git) gnuradio 3.7.4
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
Using Volk machine: sse4_2_64_orc
gr::log :INFO: audio_osx_sink0 - 

Using output audio device 'Ingebouwde uitgang'.
  ... which is the current default output audio device.
  Changing the default output audio device in the System Preferences will 
  result in changing it here, too (with an internal reconfiguration).

Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
BookmarksFile is /Users/loewal/.config/gqrx/bookmarks.csv
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
Could not resolve property : pattern10600
gr-osmosdr v0.1.1-13-g58d95b51 (0.1.2git) gnuradio 3.7.4
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace 

FATAL: No FUNcube Dongle found.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

csete, 2016-04-03:
Ok, I think I I found out what is wrong, see issue #357 

Closing this issue as a duplicate of #357

closed 2016-04-03 by csete

loewal, 2016-04-03:
Ok!!

Are you going to recompile for FCD Pro+ Alexandru?

Have a nice day!
Loe

csete, 2016-04-03:
I have just released a new OS X bundle with a quick fix.