gqrx-sdr/gqrx #337: Dock Widget Registry & widget communication?

ajmas, 2016-01-28:
Looking at the Gqrx implementation, I was wondering whether you think the following architecture changes could be of interest:
  - Using a dock widget registry for registering the dock widgets
  - Using some sort of event stream for having widgets talk to each other

The idea is to make it so the mainwindow doesn't need to have the dock widgets hard coded and to eventually make it easier to add new widgets, as the need presents itself.

csete, 2016-10-14:
Sounds like an interesting idea but it is too theoretical for me. I think I would need to see a practical demo for being able to see the need for this. So if you don't mind, I will close this issue.

ajmas, 2016-10-14:
sounds fine to me. I can reopen when I have something more concrete?

csete, 2016-10-14:
You are welcome to post a new issue if you have something implemented; however, my own priorities are to make gqrx stable and low maintenance so that I can move on to other projects (which I've been trying to do for several years now).

closed 2016-10-14 by csete