gqrx-sdr/gqrx #338: Display Frequency Allocation for displayed frequency band

ajmas, 2016-01-28:
Provide a widget for displaying the frequency allocations for the displayed band.

I have a branch with the prototype here: https://github.com/ajmas/gqrx/tree/allocations-view

A few things need to investigated:
  - is there interest from the general community or the author for having this in Gqrx?
  - is a dependency on a website for the information acceptable?
  - can this be made to work without needing to depend on a website?
  - how to ensure data freshness

This feature depends on https://github.com/ajmas/EarthFrequencies for now.

csete, 2016-01-28:
Hi,

I wish I had better time to provide feedback on this but right now I am swamped with help requests and bug fixing. So, I hope somebody else can give feedback.

I can however tell you already now, that I am quite unlikely to accept QtWebkit in as a new dependency. It is a huge library (40 MB) that would need to be bundled in e.g. the OS X versions of gqrx. Gqrx is already loaded with tons of dependencies, so we should work towards shrinking it, not increasing it.

ajmas, 2016-01-28:
Fair enough. Another approach I'll look at is simply making JSON requests and rendering using a non-webkit widget. It would probably also make it easier to keep the files locally.

ajmas, 2016-02-04:
I dropped the requirement for Webkit and I am now using a TreeWidget. This also makes it easier to store the data files locally. I have checked it into the same branch, but there is probably more work to do and possibly improvements.

The more I work on this, the more I wonder whether this widget could turn into a "Band Information" widget, indicating the band names and other textual data, though one thing at a time.

Analias, 2016-04-02:
@ajmas Do you have a screenshot of this in action?  I'm curious if some of the behavior can be factored out  to allow supporting DX spots, quick tagging of frequencies, etc. Support for overlaying multiple sources of information on the FFT would be quite useful.

alexf91, 2016-04-02:
I'm not ajmas, but here's a screenshot anyways:
[image]

It's implemented as a seperate Dock Widget, so waterfall overlays would need additional work.

ajmas, 2016-04-02:
Thanks @alexf91 for the screenshot . The initial implementation was more a proof of concept and to test the waters of interest, though still designed to be useable. One thing in this current design is that the widget could be used to provide other data, such as band name and other band specific info.

There is a current limitation in that band displayed is a single point, since I hand't worked out how to make it reflect the band the plot was showing.

One of the things I wanted to keep in mind is that anything depending on this functionality needs to continue working even if the data is not available, such as if the server offline or client has no internet connection.

The data is being pulled in as JSON format and I have tried to make it so that the data can be read from an http:// or file:// location. The default URL is hard-coded, but can be overridden in the settings file (manual edit as this point).

As a point of note, I put together earthfrequencies.org and it is very much a one person show at this point, so more effort will be needed to improve the data quality and coverage. Being an open data project, I welcome any collaboration or contributions. The repo for reference is at https://github.com/EarthFrequencies

ajmas, 2016-04-02:
Just to add the waterfall overlay would need additional work as @alexf91 indicates. For example, moving the request into an common class that can be shared amongst view needing it, to avoid code duplication.

argilo, 2020-11-28:
A band plan was added in #767, and that's available in Gqrx 2.14.

The table of allocations there could still use some work.

I'll close this issue off since the feature has been implemented.

closed 2020-11-28 by argilo