gqrx-sdr/gqrx #339: Feature requests:  Screen saver on/off, Upconverter on/off

ntoulasd, 2016-02-02:
I would like to suggest two check buttons.
One, to disable system screen saver and gqrx running and always visible. Something like this has many video players, while watching a movie.
Second, a button to enable/disable upconverter offset. This is very useful because every time we connect or disconnect a upconverter now. Its necessary to change the offset number but typing 0 or setting the prefix number.

csete, 2016-02-02:
I tried the screensaver before, but I have no idea how to do it. I think on linux it is possible to disable the screensaver using xset and so one could start gqrx from a script surrounded by xset, but I did not have much luck with it on xubuntu.

I will consider adding a button for LO on/off in a future release, in the meanwhile, have you tried using different configuration files?

ntoulasd, 2016-02-03:
I dont know about QT but I search and found this. It may helps
http://qt.apidoc.info/5.0.2/qtmultimedia/qml-qtmultimedia5-video.html#screen-saver

I switch between my two rtl-sdr on the fly. I have one with upconverter for HF and the other for vhf and above. So an on/off switch would be great.

Also I found a bug. In Configure I/O devices if I change LNB LO value and press ok.
The value change back to whatever value has at Input controls LNB LO
[image]

ntoulasd, 2016-02-03:
Maybe a device profile is a better idea. Because ppm has to change also between two devices.

csete, 2016-02-03:
Yes, if you use different hardware devices you should definitely try using different configurations. You can even load a specific configuration from the command line using

`gqrx -c abcd.conf`

so you can create shortcuts for different devices.

csete, 2017-08-26:
Regarding screensaver, it seems on linux we could use the xdg-screensaver command.

csete, 2017-08-26:
On Mac OS X we can use UpdateSystemActivity:
https://developer.apple.com/library/content/qa/qa1160/_index.html