gqrx-sdr/gqrx #361: Mac OS X build breaking if homebrew Pulseaudio installed

Analias, 2016-04-14:
As of commit 529fb14 the build for Mac OS X is failing if the homebrew package for Pulseaudio is installed.  When I run "make" it fails with:

[ 79%] Building CXX object src/CMakeFiles/gqrx.dir/qtgui/dockrxopt.cpp.o
[ 80%] Building CXX object src/CMakeFiles/gqrx.dir/qtgui/freqctrl.cpp.o
[ 82%] Building CXX object src/CMakeFiles/gqrx.dir/qtgui/ioconfig.cpp.o
/Users/pascal/Ham_Radio/Projects/n5fpp/gqrx/src/qtgui/ioconfig.cpp:141:18: error: use of undeclared identifier 'devices'
    outDevList = devices.get_output_devices();
                 ^
/Users/pascal/Ham_Radio/Projects/n5fpp/gqrx/src/qtgui/ioconfig.cpp:185:5: error: invalid use of member 'inDevList' in static member function
    inDevList = devices.get_input_devices();
    ^~~~~~~~~
/Users/pascal/Ham_Radio/Projects/n5fpp/gqrx/src/qtgui/ioconfig.cpp:188:21: error: invalid use of member 'inDevList' in static member function
    for (i = 0; i < inDevList.size(); i++)
                    ^~~~~~~~~
/Users/pascal/Ham_Radio/Projects/n5fpp/gqrx/src/qtgui/ioconfig.cpp:190:20: error: invalid use of member 'inDevList' in static member function
        this_dev = inDevList[i].get_name();
                   ^~~~~~~~~
4 errors generated.
make[2]: *** [src/CMakeFiles/gqrx.dir/qtgui/ioconfig.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
make: *** [all] Error 2

Looking at the code it looks like the declaration for "devices" was deleted with commit 529fb14, but still being used if WITH_PULSEAUDIO and GQRX_OS_MACX are both defined.  I'm not sure the of the original intent of the update, so I'm sure the best way to resolve the problem.

csete, 2016-04-14:
Surely it is not commit 529fb14  that causes trouble bu the one before it 04034816c5d4e591b5f0498ee7010f0db8e17e97 , no?

I will take a look at it when I get some time; however, I am really surprised that it has ever worked because I have always considered pulseaudio to be a linux only option. Out of curiosity, what on Earth compels you to use pulseaudio on OS X?

Analias, 2016-04-14:
Why was I using Pulseaudio on the Mac? Good question, I don't remember the
specific reason. Either there was another package that used it, or I was
trying to get Jack working for routing. Unfortunately, I did not write it
down in my notes. I will try removing the dependency tonight. I only
reported the problem because I stumbled over it.

As for the commit number. The build starts to fail after I do a 'git
checkout 529fb14' to set my position in the history. I maybe using the
wrong method to do this, but it seems to do what I want. I did think it was
odd when I look at the diff between 529fb14 and 0403481 that there was more
than displayed by the commit history. Again I maybe doing it wrong.

csete, 2016-04-14:
Ok, thanks for the info.

I can imagine that if pulseaudio is installed and detected by cmake or qmake, then it will be used because we assume in the build scripts that it is only available on linux. If you want to try you can force gr-audio which should then use the native OS X audio interface. Let me know if you need info about how to do that.

Analias, 2016-04-14:
I do need more info to force ignoring Pulseaudio.  I tried setting the audio backend to gr-audio, and setting WITH_PULSEAUDIO=false via cmake.  The preprocessor symbol WITH_PULSEAUDIO was still getting set in the code and causing the error to occur.  If you know the magic incantation for cmake arguments, please let me know.

csete, 2016-04-14:
Yeah, I was too quick making that suggestion. It seems the options are only available on Linux, so it is actually very strange that you can build using pulseaudio on OS X.

Do you see the message "Gqrx configured with pulseaudio backend" when you use qmake?
If you do, the somehow Qt is not aware of that it is OS X or perhaps I am using an obsolete switch on the gqrx.pro file.

Analias, 2016-04-14:
I'll check tonight when I get home.  I don't remember seeing the message, but that doesn't mean it wasn't there.

a1k0n, 2016-04-15:
I also ran into this and quickly patched it... the commit which made the initialization function static broke the mac build. Here's a patch which gets the build working (I'll make a pull request later on):

```
diff --git a/src/qtgui/ioconfig.cpp b/src/qtgui/ioconfig.cpp
index 96b3b18..97eae13 100644
--- a/src/qtgui/ioconfig.cpp
+++ b/src/qtgui/ioconfig.cpp
@@ -137,7 +137,7 @@ CIoConfig::CIoConfig(QSettings *settings, std::map<QString, QVariant> &devList,

 #elif defined(GQRX_OS_MACX)
     // get list of output devices
-    // (already defined) osxaudio_device_list devices;
+    osxaudio_device_list devices;
     outDevList = devices.get_output_devices();

     qDebug() << __FUNCTION__ << ": Available output devices:";
@@ -182,7 +182,7 @@ void CIoConfig::getDeviceList(std::map<QString, QVariant> &devList)
     // automatic discovery of FCD does not work on Mac
     // so we do it ourselves
     osxaudio_device_list devices;
-    inDevList = devices.get_input_devices();
+    vector<osxaudio_device> inDevList = devices.get_input_devices();

     string this_dev;
     for (i = 0; i < inDevList.size(); i++)
diff --git a/src/qtgui/ioconfig.h b/src/qtgui/ioconfig.h
index cfb75e7..912f0b4 100644
--- a/src/qtgui/ioconfig.h
+++ b/src/qtgui/ioconfig.h
@@ -70,7 +70,6 @@ private:
     vector<pa_device> outDevList;
 #elif defined(GQRX_OS_MACX)
     vector<osxaudio_device> outDevList;
-    vector<osxaudio_device> inDevList;
 #endif

 };
```

csete, 2016-04-15:
Thanks but pull request will not make fixing this any faster. I just have to figure this one out for myself  and ensure that it works on all platforms.

Analias, 2016-04-15:
Andy, I'm glad I'm not the only one hit by this. That tells me the problem
is less likely with what I have installed on my Mac.  I will apply your fix
to my own fork for the short term until Alex resolves the issue more
permanently. I will at least be able to keep up with other changes off the
upstream until then.  Thanks.

csete, 2016-04-15:
@Analias I have applied the patch from @a1k0n  but does it fix your build issues?

Analias, 2016-04-16:
@csete Yep, that patch fixed the immediate problem. I can build again.  That should give you a bit of room to find the final solution you want.  -- Thanks.

csete, 2016-04-16:
The solution I want is this patch, I just had to review the issue in context. What I wanted to know if this also fixes all your build problems, if yes, we can close this issue.

closed 2018-03-18 by csete